nasa/nos3 #230: Wiki Update for Docker

jlucas9, 2023-11-30:
Update the wiki documentation to be specific to docker in preparation for the next release.

kevincbruce, 2024-02-23:
Cloned Branch nos3#230-wiki-update-for-docker to work on this issue and Igniter Documentation.

rjbrown2, 2024-02-29:
On the Wiki front page, there are some link naming mismatches:
Navigation links in the middle of the page do not match the naming on the right side bar.  For example: Architecture vs NOS3 Architecture.  They are close enough to not cause confusion, but they should probably match.

jlucas9, 2024-02-29:
Can we also resolve https://github.com/nasa/nos3/issues/136 as part of this effort?

kevincbruce, 2024-02-29:
> On the Wiki front page, there are some link naming mismatches: Navigation links in the middle of the page do not match the naming on the right side bar. For example: Architecture vs NOS3 Architecture. They are close enough to not cause confusion, but they should probably match.

Should have fixed that. Those were how the names originally were before I made edits, but should match now.

kevincbruce, 2024-03-01:
Addressed most of #136 today, so that is ready for review.

kevincbruce, 2024-03-04:
Finished addressing #136 (both in code and in documentations), and fixed the description and pictures of enabling TO for Radio. Starting Architecture updates tomorrow afternoon.

kevincbruce, 2024-03-11:
There is a fair bit of copy and paste, but I updated the Ground Software wiki to reflect the two ground systems, show Radio TO setup on both, and show how to use the Master Config to toggle them

jlucas9, 2024-04-25:
Closing in favor of new issues as things continue to be updated

closed 2024-04-25 by jlucas9