nasa/nos3 #231: Arm64 Linux support

AdilHydari, 2023-12-05:
As an Asahi Linux user on Apple Silicon, I am facing the challenge of running Nos3 on ARM64 due to the lack of native support. While FEX support is still under development and QEMU is not performant enough, using a hypervisor is not a viable option for me. Given the potential performance impact, is it worthwhile to invest the effort in adding native ARM64 support to Nos3? Additionally, both the Docker image and Vagrant box require nos3_engine_x86.deb/rpm, which creates a conflict when attempting to deploy locally. How can this dependency issue be addressed?

Thanks in advance.

jlucas9, 2024-03-19:
Hi @AdilHydari,
I'm working to get this on our development roadmap to ensure support on ARM64.
Apologies for the delay in getting it going.

AdilHydari, 2024-03-19:
Not to much of an issue, I understand the user-base for this sort of thing is quite small, especially for ARM64 linux users. Let me know if I can do anything to help out, I would consider myself _alright_ at the cFS and nos3 architecture.

jlucas9, 2024-05-15:
While I get this sorted properly, note that some users have reported leveraging the existing VM with https://mac.getutm.app/

Two changes they reported making to settings - set CPU to Default and Display adapter to virtio-gpu-gl-pci.

I haven't tried this personally, but wanted to capture it as an option.

sdunlap-afit, 2024-06-14:
I don't have a Mac to test on, but I can confirm that this is working on a Raspberry Pi 4 running Raspberry Pi OS (64-bit). I should say that a Raspberry Pi is extremely under-powered for running all of NOS3 (actual speed-up = 0.25) Dependencies for fresh Raspberry Pi:

```bash
sudo apt install -y docker.io gnome-terminal
sudo usermod -aG docker $USER
sudo reboot
```

The only issue I've found is with `scripts/prepare.sh`, in the `Prepare Igniter...` section. The script is trying to install `pyside6` and `xmltodict` using `pip3`, but it fails because the environment is externally managed. I got around it using `python -m venv env; source env/bin/activate`. I also had to install one more dependency using `sudo apt install libxcb-cursor0`. Any reason `cfg_gui_main.py` can't be run in a Docker container instead of on the host?

szemerick, 2024-06-16:
@sdunlap-afit Thanks for the notes! You beat me to it as I was trying the same thing.  I'm just curious how the performance will be on a rPI 5, which I'm going to try this week.

To add to your notes I also did: `sudo apt-get install python3-full`

In the `scripts/prepare.sh` change the path to the one where you activated the venv.

jlucas9, 2024-06-18:
Created a follow on issue to containerize igniter - https://github.com/nasa/nos3/issues/319

Thanks for testing this out and providing the great notes!

jlucas9, 2024-06-19:
Merged to `dev`

closed 2024-06-19 by jlucas9