nasa/nos3 #235: NOS Engine Server Terminal errors and FSW node

StefanoSWE, 2023-12-20:
Hello!
I'm a new user of NOS3 and I'd like to have support on the following situation.

Would you mind explaining me why NOS Engine Server Terminal is reporting the following errors?

[image]

Moreover, why isn't _fsw_ present on every bus but only on some busses? E.g. _fsw_ present on _usart_2_ but not on _usart_1_.

[image]

I'm using the main branch.

Thank you so much for your support.

jlucas9, 2024-01-04:
Hi @StefanoSWE,
I can confirm on the NOS3 main branch (release 1.6.2) I get the following on initial boot:
[image]

By default only the sample component is enabled by default. The other components need enabled manually using the OpenC3 command sender which is why only some interfaces show multiple connections:
[image]

Once that is complete the NOS Engine Server should report the following:
[image]

The errors reported by NOS Engine will hopefully be resolved in the next update to Engine, but can be safely ignored for now. Please post any other questions you have and I'll try to get back to you quickly now that I've returned from a long holiday.

jlucas9, 2024-03-19:
Closing for now, please reopen if more questions arise!

closed 2024-03-19 by jlucas9