nasa/nos3 #244: NOS3 bus connection capabilities

IKuraudo, 2024-01-15:
Hello everyone,

I hope this message finds you well. I'm currently working on establishing communication between two hardware models using NOS connections through UART or I2C.

I've made progress with the UART interface, leveraging the provided sample that covers most of the UART functionality. However, I'm encountering an issue when trying to send a custom 12-byte message to a connected UART bus (usart_5) from my sender hardware model.

Here's a snippet of the code I'm using, based on the standard <Uart/Client/Uart.hpp> header:
```
std::vector<uint8_t> out_data; 	//declare output data vector
std::vector<uint8_t> in_data;	//declare input data vector

_uart_connection->open(node_port);  // open uart connection
_uart_connection->write(&out_data[0], out_data.size());     // write to uart port
_uart_connection->close();      // close port

to_uart(var_to_send,out_data);	//create custom message

in_data.resize(12, 0x00);   // resize input to correct size
_uart_connection->open(node_port);  // open port
_uart_connection->read(&in_data[0], in_data.size());  // read from port
_uart_connection->close();  // close port
```

Despite following this approach, I'm not receiving the intended message back from the bus. Instead, I'm getting an empty array. I wanted to confirm if this is the correct method for handling UART communication within NOS3. Also, I'm curious if NOS3's connections are designed to facilitate communication between subsystems. Could you guide me on how the UART connection in the NOS engine works? For instance, where can I find the <Uart/Client/Uart.hpp> header used for this connection?

I appreciate your assistance, and I'm eager to hear your insights on this matter.

Thank you!

jlucas9, 2024-01-29:
Hi @IKuraudo,
Apologies for the delay in getting to this issue. Required a little bit of work to provide a proof of concept to reference. The `nos3#244-sim-to-sim` branch includes updates to the checkout script, sample simulator, and a new subsim. 

How I build and launched:
* Assuming using the latest `nos3_20231101` base box
* `git checkout nos3#244-sim-to-sim`
* `make clean config sim`
  * Cleans up existing code, generate the configurations for FSW / GSW / Sims, build the sims
* `make debug`
  * `cd ./components/sample/support`
  * `mkdir build`
  * `cd build`
  * `cmake .. -DTGTNAME=cpu1`
  * `make`
  * `exit`
* `make checkout`

How I tested:
* Move windows around to see all four easily
* Use `Sample Checkout` to request HK message
  * `h` in terminal
* Confirm success
* Use `Sample Checkout` to send NOOP
  * `n` in terminal
* Confirm success in all three terminals: `Sample Checkout`, `Sample Sim`, and `Subsim`

[image]

Let me know if you have any questions or issues in replicating this or how things work!

jlucas9, 2024-02-07:
Hi @IKuraudo,

I just went with the sample checkout initially as it's easier to focus on just the pieces I've modified without other interactions. The same sample_device.c code is utilized by the application. I've updated the scripts to allow running `make all` followed by `make launch` after waiting for the COSMOS plugins to finish loading:
[image]

In your code are you attempting to communicate via the same usart bus for both the FSW<->SampleSim and SampleSim<->SubSim? If so that's likely the issue as USART is designed to be point to point. For example in NOS Engine Server with the example above querying the data nodes (2) I receive:
[image]

Let me know if you have any other questions and have a great day!

IKuraudo, 2024-02-12:
Hello @jlucas9,
thank you for your continued support. Currently, I am indeed trying to use the same port for the Sender<->Receiver and Receiver<->fsw connection, which is the one failing. However, it would be required to connect both models to the fsw in my case. 
In total, there should be three uart connections like (1) Sender<->Receiver, (2) Sender<->fsw, and (3) Receiver<->fsw. 
Is this possible within the nos3 and how would I need to configure the uart_command_callback to be able to facilitate the communication?
Again, thank you and have a great day!

jlucas9, 2024-02-29:
Hi @IKuraudo,
I think to keep using a UART you'd need to have multiple lines for each communication path.
* usart_20
  * (1) Sender <-> Receiver
* usart_21
  * (2) Sender <-> fsw
* usart_22
  * (3) Receiver <-> fsw

Looking at it the other way:
* Sender
  * usart_20, node 20
  * usart_21, node 21
* Receiver
  * usart_20, **node 21**
  * usart_21, node 21
* FSW
  * usart_21, node fsw
    * Send data to Sender
  * usart_22, node fsw
    * Read data from Receiver

Note that the no two pieces on the shared usart can have the same node value. Let me know if this continues to give you issues! NOS Engine has a few oddities like this I need to better document.

jlucas9, 2024-03-19:
Closing for now, please re-open if new issues arise!

closed 2024-03-19 by jlucas9