nasa/nos3 #248: It is possible to reduce the virtual size from 64GB to 20GB?

AntoniMorote, 2024-01-18:
I am doing a project and I would need to reduce the virtual disk space to be able to upload it to cleura without it taking too long. And I don't see how I can reduce the size of the disk, if anyone knows it would be a great favor for me.

jlucas9, 2024-01-18:
Hi @AntoniMorote,

A couple quick options:
* VirtualBox OVF export
  * Assuming you have the Vagrant box working already locally, in VirtualBox you can right click the VM and export to OVF
  * The exported OVF size should be < 20GB if it's a fresh VM (no additional installed packages, built stuff, docker containers, etc.)
  * https://support.cloudshare.com/hc/en-us/articles/360051851812-Importing-VMs-with-the-Wizard
* Provisioning yourself
  * Use a provided Ubuntu 22.04 image (or manually install it) then use the deployment repository to install the necessary packages directly on it
  * If using NOS3 `dev` branch start with https://github.com/nasa-itc/deployment/tree/20231101
  * If using NOS3 `main` branch start with https://github.com/nasa-itc/deployment/tree/20230718
  * The NOS3 deployment repository has a `deploy-local-env.sh` script that ought to work to install the required packages or you can handle it yourself apt installing packages listed here `https://github.com/nasa-itc/deployment/blob/main/ansible/roles/nos3-software-env/vars/Debian.yml`

Let me know if this doesn't work or if you have any other questions!

AntoniMorote, 2024-01-23:
I have done it all. Now I go into the mission folder, I do make and I get this error
nos3@nos3:~/deployment/mission$ make
make fsw
make[1]: se entra en el directorio '/home/nos3/deployment/mission'
make fsw-prep
make[2]: se entra en el directorio '/home/nos3/deployment/mission'
mkdir -p /home/nos3/deployment/mission/fsw/build
cd /home/nos3/deployment/mission/fsw/build && cmake -DCMAKE_INSTALL_PREFIX=exe -DCMAKE_BUILD_TYPE=debug ../cfe
CMake Deprecation Warning at CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
-- OMIT_DEPRECATED=false: Deprecated elements included in build
-- Search path for modules: apps;libs;psp/fsw/modules;cfe/modules
-- Module 'cfs_lib' found at /home/nos3/deployment/mission/fsw/apps/cfs_lib
-- Module 'hwlib' found at /home/nos3/deployment/mission/fsw/apps/hwlib
-- Module 'io_lib' found at /home/nos3/deployment/mission/fsw/apps/io_lib
-- Module 'ci' found at /home/nos3/deployment/mission/fsw/apps/ci
-- Module 'lc' found at /home/nos3/deployment/mission/fsw/apps/lc
-- Module 'sc' found at /home/nos3/deployment/mission/fsw/apps/sc
-- Module 'sch' found at /home/nos3/deployment/mission/fsw/apps/sch
-- Module 'to' found at /home/nos3/deployment/mission/fsw/apps/to
-- Module 'arducam' found at /home/nos3/deployment/mission/fsw/apps/arducam
-- Module 'generic_reaction_wheel' found at /home/nos3/deployment/mission/fsw/apps/generic_reaction_wheel
-- Module 'novatel_oem615' found at /home/nos3/deployment/mission/fsw/apps/novatel_oem615
-- Module 'sample' found at /home/nos3/deployment/mission/fsw/apps/sample
-- Module 'cfe-core' found at /home/nos3/deployment/mission/fsw/cfe/fsw/cfe-core
-- Module 'osal' found at /home/nos3/deployment/mission/fsw/osal
-- Module 'psp' found at /home/nos3/deployment/mission/fsw/psp
-- Generated cfe_mission_cfg.h from /home/nos3/deployment/mission/fsw/nos3_defs configuration
-- Generated cfe_perfids.h from /home/nos3/deployment/mission/fsw/nos3_defs configuration
CMake Deprecation Warning at /home/nos3/deployment/mission/fsw/tools/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
-- Configuring for system arch: cpu1/default_cpu1
CMake Deprecation Warning at CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.
  Update the VERSION argument <min> value or use a ...<max> suffix to tell
 CMake that the project does not need compatibility with older versions.
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - failed

-- Check for working C compiler: /usr/bin/gcc

-- Check for working C compiler: /usr/bin/gcc - broken

-- Configuring incomplete, errors occurred!

See also "/home/nos3/deployment/mission/fsw/build/cpu1/default_cpu1/CMakeFiles/CMakeOutput.log".

See also "/home/nos3/deployment/mission/fsw/build/cpu1/default_cpu1/CMakeFiles/CMakeError.log".

CMake Error at /usr/share/cmake-3.22/Modules/CMakeTestCCompiler.cmake:69 (message):

  The C compiler
    "/usr/bin/gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/nos3/deployment/mission/fsw/build/cpu1/default_cpu1/CMakeFiles/CMakeTmp
        Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_ffbf6/fast && gmake[3]: se entra en el directorio '/home/nos3/deployment/mission/fsw/build/cpu1/default_cpu1/CMakeFiles/CMakeTmp'

[... 43 lines cut ...]
Call Stack (most recent call first):

  CMakeLists.txt:63 (project)

CMake Error at cmake/mission_build.cmake:353 (message):

  Failed to configure cpu1_default_cpu1

Call Stack (most recent call first):

  CMakeLists.txt:123 (process_arch)

-- Configuring incomplete, errors occurred!

See also "/home/nos3/deployment/mission/fsw/build/CMakeFiles/CMakeOutput.log".

make[2]: *** [Makefile:53: fsw-prep] Error 1

make[2]: se sale del directorio '/home/nos3/deployment/mission'

make[1]: *** [Makefile:48: fsw] Error 2

make[1]: se sale del directorio '/home/nos3/deployment/mission'

make: *** [Makefile:41: all] Error 2

jlucas9, 2024-02-29:
Hi @AntoniMorote,
I wonder if maybe there is a bug in the deployment/mission version of the code. Can you try cloning the main branch of `nos3` in the VM and seeing if things are happier for you?

I've seen that `gcc --isbroken` previously and will see if I can dig up any notes as to why that my be happening in the mean time.

jlucas9, 2024-03-19:
Closing for now, please re-open if more assistance is needed!

closed 2024-03-19 by jlucas9