nasa/nos3 #255: NOS3 Local Issue(s)

SpaceCyberCoffee, 2024-02-09:
Hi,

NOS3 runs fine with the provided VMs, even with the latest Vagrant (2.4.x) and Docker (>=25.x) versions.

However I have problems with NOS3 being run locally within Ubuntu 22.04 LTS.

Even though the `make prep` and the `make` report no issues, after doing the `make launch` the terminal tabs and windows are opened, COSMOS is running, but I have the following problems:

NOS3 Flight Software terminal window shows this error:

> EVS Port1 42/1/SAMPLE 32: SAMPLE: Request device data reported error -1

In Cosmos, “Debug” and “Radio” interfaces are stuck at “Attempting” and the “NOS3 Time Driver” stops at tick = 920

And the three 42 windows are opened, but they are all black.

Any workarounds for this? 
Or I am missing something?

Thank you in advance for your support and for your amazing work with NOS3!

jlucas9, 2024-02-15:
Hi @SpaceCyberCoffee ,

Thanks for submitting the issue. I can confirm a fresh Ubuntu 22.04 results in similar problems for me:
[image]

Appears the bug was in the sample simulator. It was crashing then leaving the entire system to hang. Can you give the `nos3#255-Ubuntu2204` branch a try and see if it fixes it for you?

SpaceCyberCoffee, 2024-02-15:
Hi @jlucas9 thank you for your Ubuntu2204 branch!
I have tested it and it solved most of the issues I had before!
I am only left with two problems: Debug and Radio interfaces in COSMOS are still stuck at "Attempting", but both 42 and the Flight Software are running!

The error messages for "Radio" and "Debug" are the following (from COSMOS Log Messages):
```
RADIO: Connection Failed: SocketError : Hostname not known: radio_sim
DEBUG: Connection Failed: SocketError : Hostname not known: nos_fsw
```

Should I manually modify the Hostnames somewhere?

Thank you again for the dedicated branch!

jlucas9, 2024-02-15:
@SpaceCyberCoffee - I'm happy to help!
Did you install cosmos using the `make prep` or manually? 

I'd recommend doing a `make stop-gsw` followed by `rm -r ~/.nos3` and then `make prep` again just to be sure the version of OpenC3 cosmos with ports already open for you is running.

Here are the changes are from the main OpenC3 repository if you wanted to manually recreate things above:
* https://github.com/nasa-itc/openc3-nos3/commit/1b503cd57eff5361bb1182952b875a56079d48a8

SpaceCyberCoffee, 2024-02-15:
Hi again @jlucas9, thank you for your quick reply!
I initially used `make prep`, then I followed your instructions (`make stop-gsw` -> `rm -r ~/.nos3` -> `make prep`), but I still have the same issue as before.

I also did a `make clean` and did all the steps again.

However, upon the completion of the `make prep` command, the interfaces are already popping-up in Cosmos, with Radio and Debug being stuck in "Attempting".

For what it is worth, COSMOS did not prompt for a new password.

Indeed, I have noticed the following message while after doing the `make prep`:

> WARN[0000] a network with name openc3-cosmos-network exists but was not created for project "cosmos".
> Set `external: true` to use an existing network

jlucas9, 2024-02-18:
Hi @SpaceCyberCoffee,

I double checked following those same steps myself and things seem happy. Do you see any other errors reported during the `make` process to build everything?

[image]

One thought I had was maybe the groups your local user are a part of could be preventing editing the docker boxes that COSMOS runs in. Here is what I am in now:
```
nos3@nos3-VirtualBox:~/git/n0$ groups
nos3 adm cdrom sudo dip plugdev lpadmin lxd sambashare docker vboxsf
```

Note that I am running a windows host with Ubuntu 22 in a VM and my nos3 folder shared in - this required the additional `vboxsf` group.

SpaceCyberCoffee, 2024-02-20:
Hey @jlucas9 

This is the output of the `make launch`. 
No errors in the terminal are reported.

[image]

On the other hand, there is error in the NOS3 Engine Server terminal tab:
[image]

SpaceCyberCoffee, 2024-02-20:
Hey @jlucas9,
the `make launch` command doesn't return any error.
[image]

However there is the following error in the NOS3 Engine Server terminal tab:

[image]

SpaceCyberCoffee, 2024-02-21:
Hi @jlucas9, just a short update.

I deleted all the NOS3 and OpenC3 COSMOS Docker volumes, containers and images.
I deleted the ~/.nos3 folder and re-downloaded the feature branch #255 from scratch.

This time, at the end of the make prep, "Debug" and "Radio" are not within the Cosmos interfaces, that is empty (see screenshot).
[image]

But unfortunately after running the make command, that gives no errors, I am back to "Debug" and "Radio" being stuck at "Attempting", with the usual error message(s).

jlucas9, 2024-02-23:
Hi @SpaceCyberCoffee,
I get similar errors in NOS Engine but not identical to what you reported:

[image]

Does time appear to tic forward for you after you `make launch`?

You should be able to see this in the NOS3 Time Driver tab and in the 42 windows:
[image]

Assuming that's happy - I'm thinking maybe the network connections between the docker containers is mad. While everything is still running after `make launch`, can you `docker network ls`?

[image]

Additionally, I receive the following for a `docker network inspect nos3_sc_1`

[image]

jlucas9, 2024-02-29:
Hi @SpaceCyberCoffee,
This seems to be the same thing captured in #242. The quick fix to get you fully functional is to edit the [./scripts/docker_launch.sh Line 84](https://github.com/nasa/nos3/blob/46e9ed27357891342179430682dbc917ef967b60/scripts/docker_launch.sh#L84) to use the `-` instead of the `_` when it's doing the network assignments. 
Hopefully this does it for you!

SpaceCyberCoffee, 2024-02-29:
Hi @jlucas9,
indeed it did the trick!
I had tried before but I hadn't realize that there were 2 underscores `cosmos**_**openc3-operator**_**1`, and not only the one before the "1"!

I am happy to report that now NOS3 works flawlessly on Ubuntu 22.04.4 LTS, with Docker version 25.0.3.

Unrelated note: I have also been playing around with the latest version of `42`.
I saw that you have are planning to implement it (#207 ) in the future release.

I am happy to test it if you want and/or to share my attempt!

jlucas9, 2024-03-19:
Just merged in #207 to the latest `dev` branch @SpaceCyberCoffee if you wanted to give it a shot!

You'll need to `make real-clean prep all` to:
* real-clean
  * Clean up all the old stuff hanging around
* prep
  * Download the latest and greatest
* all
  * Build everything

Please feel free to file more issues as you find them!

closed 2024-03-19 by jlucas9

SpaceCyberCoffee, 2024-03-27:
Hi @jlucas9,
I pulled the new `dev`, then edited the `scripts/docker_launch.sh` file to have `cosmos-openc3-operator-1`, and I followed your instructions, but now I get these errors, at the end of "Install the project.." part:

```
make[1]: Leaving directory '/home/user-dir/nos3_local/nos3'
make gsw
make[1]: Entering directory '/home/user-dir/nos3_local/nos3'
./cfg/build/gsw_build.sh
make[1]: ./cfg/build/gsw_build.sh: Permission denied
make[1]: *** [Makefile:88: gsw] Error 127
make[1]: Leaving directory '/home/user-dir/nos3_local/nos3'
make: *** [Makefile:46: all] Error 2
```