nasa/nos3 #260: Creating additional subsystems

ullixes, 2024-02-26:
Dear NOS3-Team,

I installed NOS3 and find it super interesting! Looks very good!

However, I wanted to start the development of my own subsystem and I am not able to get it to run. I am using the dev-branch. So far, I tried with my own code for the subsystem hardware model, but I was also trying to just “copy” the sample subsystem. Both times I followed the instructions with the generate_template.sh and I also found the configure.py-script and added the new subsystem and also everywhere where I could find that the “sample” subsystem was mentioned, e.g. in docker_launch.sh, docker.yml, the cfg folder, nos3-defs folder and tables folder, etc. I edited the uart ports, 42 port, address for COSMOS in the corresponding files (newsim_msgids.h, newsim_platform_cfg.h, hardwaremodel.cpp etc.). After doing “make prep”, “make config" and "make -j4", sometimes the new subsystem just crashes, sometimes the uart-connection is not established  between cFS and the hardware model (when checking the "Nos Engine Server" tap), sometimes cFS gets kind of stuck.

Since it is not possible for me to understand what is wrong, I wanted to ask you if you could check if your proposed workflow for creating a new subsystem is still working with the dev-branch, and if this is the case, if you could provide me a corresponding repository with added new subsystem, so that I can look for my own errors?

Also, looks like the latest version of the dev-branch with the new subsystem “sys” cannot be cloned. I get the following error:

Cloning into '/my/folder/nos3/components/syn'... ssh: connect to host github.com port 22: Connection timed out fatal: Could not read from remote repository. Please make sure you have the correct access rights and the repository exists. fatal: clone of 'git@github.com:nasa-itc/syn.git' into submodule path '/my/folder/nos3/components/syn' failed Failed to clone 'components/syn' a second time, aborting

Thank you very much!

jlucas9, 2024-02-29:
Hi @ullixes,
I just noticed that the latest `syn` merge into the `dev` branch uses a git link instead of https which could be causing that issue. I'll file an issue here in a moment to get that resolved and fix another oddity I'm seeing with it now.

As far as creating another subsystem, the issue maybe that the new component you've created is clashing with another assigned uart port. Have you updated the values in the [sample_hardware_model.cpp](https://github.com/nasa-itc/sample/blob/744d049da704252d8615e6f0e96dba77a25f0443/sim/src/sample_hardware_model.cpp#L23) for bus_name and node_port? Note you'll also need to do this for the FSW at [sample_platform_cfg.h](https://github.com/nasa-itc/sample/blob/744d049da704252d8615e6f0e96dba77a25f0443/fsw/platform_inc/sample_platform_cfg.h#L21). 

To more easily see what is causing the crash you can modify the [./scripts/docker_launch.sh](https://github.com/nasa/nos3/blob/46e9ed27357891342179430682dbc917ef967b60/scripts/docker_launch.sh#L100) to hold the terminal open instead of close after an error occurs. You'll need to create the new gnome-terminal profile manually first by:

* Going to preferences
* Cloning the existing profile
  * [image]
* Renaming to `KeepOpen`
* Under the `Command` tab, change the When command exists to `Hold the Terminal Open`
  * 
[image]

Let me know as you have issues or if the above doesn't work as I'd be happy to keep trying to debug with you!

ullixes, 2024-03-01:
Hi @jlucas9,

thank you very much for your response and for your offer to help debugging! I will use the the new terminal settings and check if I can find out something.

Yes, I changed the uart port settings for the new subsystem including cFS app in newsim_platform_cfg.h. But I have a question regarding the ./cfg/nos3_defs/tables folder. Which files are mandatory to change for a new subsystem? For example, the lines for the sample subsystem in "sch_def_schtbl.c" and "sch_def_msgtbl.c" are only for the scheduler app of cFS to define the constant sending of TM, right?

Another question that I have is that I added some code to the "generic_eps_hardware_model.cpp", which is a run()-function including some text output. But it looks like when the eps is started, the constructor is not fully executed because I do not get the "Generic_epsHardwareModel::Generic_epsHardwareModel:  Construction complete." output line. Instead the i2c interface starts to produce its output. Also, I do not get any output I defined in the added run()-function via sim_logger. Do you know the reason for this?

Thanks again!

jlucas9, 2024-03-05:
Hi @ullixes,

That's correct - the `./cfg/nos3_defs/tables` folder only needs updated if you'd like to have SCH send period requests to your new application for telemetry or to do scheduled work. Additionally the TO tables can be updated to send desired messages through the radio (standard TO or Telemetry Output) or the debug port (TO_Lab). 

Do you happen to have a fork of NOS3 I could get access to in order to debug further and inspect the code some? In the latest `dev` branch, I do see (if you scroll way back up quickly) the successful constructor message before you get flooded with the battery update prints.

[image]

ullixes, 2024-03-08:
Hi @jlucas9,

thank you for your response. Unfortunately, for now I do not have a NOS3 fork available. I attached the files, with the main changes and hopefully I did not forget one. The new subsystem I added is called "newsim". Apologies for the circumstances. If you could check these files would be very helpful for me. When I compile and run the code the flight software gets stuck, see screenshots.

Regarding the other problem with the EPS, I will check again and get back to you.

Many thanks for your support! 

[image]
[image]
[image]
 
[newsim.zip](https://github.com/nasa/nos3/files/14539016/newsim.zip)

jlucas9, 2024-06-25:
@ullixes - sorry for the delay getting back to you! 

I've created a new branch `nos3#260-newsim` where I integrated this. Some minor additions:
* Added newsim to the targets.cmake file to build the FSW side
* Added 4251 to the Inp_IPC.txt file
* Added NEWSIM to cosmos system and cmd_tlm_server
* Minor edits due to new strict build flags

[image]

Let me know if run into other issues! Feel free to keep tagging me every so often too so it doesn't fall off my radar!

jlucas9, 2024-07-23:
Closing due to age

closed 2024-07-23 by jlucas9