nasa/nos3 #267: Instructions on separating FSW and GSW onto different machines

jlucas9, 2024-03-05:


sdunlap-afit, 2024-06-14:
In case it helps someone, I've been working on getting FSW running on a Rapsberry Pi, while the rest is running on another machine. While it's not very clean, I've documented my process:

https://github.com/sdunlap-afit/nos3_hwil

Isgaroth-the-Green, 2024-07-30:
@sdunlap-afit Out of curiosity, in your instructions, is the IP address you put in the /etc/hosts file the IP address of the VM running the FSW, or the VM running everything else? Also, did you put the other containers on the host networking as well? And if not, how did those connect?

sdunlap-afit, 2024-07-30:
@Isgaroth-the-Green  The IP address in /etc/hosts is the IP of whatever host (or VM) you are running everything else on. Specifically, it is the host running the "NOS Engine Server" docker container. This way, when FSW tries to connect to "nos_engine_server" through the host interface, it will get directed to that IP.

Since all other containers are running on the same host, they can stay on the "$SC_NETNAME" network. Adding `-p 12000:12000` to "NOS Engine Server" docker run command simply allows access to that port from the host running the FSW.

Isgaroth-the-Green, 2024-07-30:
I see. You said that your instructions work for you running it on a Raspberry Pi? You have communication on the DEBUG interface of COSMOS (and, for instance, can successfully send no-op commands)?

sdunlap-afit, 2024-07-30:
Ah, I had not gotten that far; I'm still new to nos3. I played around with it a bit and did manage to get it communicating, but there are some more hoops to jump through. This is by no means a clean process, but I made a few hackish changes to get it working:

1. Add "cosmos IP" to `/etc/hosts` (same IP as `nos_engine_server`).

On the main machine: 

1. In "cfg/nos3_defs/cpu1_device_cfg.h" change `GENERIC_RADIO_CFG_FSW_IP` to the actual IP instead of the `nos_fsw` hostname.
2. In "scripts/gsw_cosmos_launch.sh" add `-p 5013:5013/udp` to the `docker run` command.

As a side note, folks may need to disable their firewall or open specific ports. For my main machine, I'm using Windows 11, WSL, and Docker Desktop.

[image]

Isgaroth-the-Green, 2024-11-07:
Closed by https://github.com/nasa/nos3/pull/375

closed 2024-11-07 by Isgaroth-the-Green