nasa/nos3 #269: Issues building outside of Vagrant environment

bjkowalski, 2024-03-14:
If a user attempts to perform `make` outside of a Vagrant environment errors will be reported and the build will fail.

Performed the following commands:
```
git clone -b dev https://github.com/nasa/nos3.git
cd nos3
git submodule update --init --recursive
make prep
make
```
The following error was generated:
```
make gsw
make[1]: Entering directory '...../nos3'
./cfg/build/gsw_build.sh
make[1]: ./cfg/build/gsw_build.sh: Permission denied
make[1]: *** [Makefile:87: gsw] Error 127
make[1]: Leaving directory '...../nos3'
make: *** [Makefile:46: all] Error 2
```
The error is generated because four files have been checked into the GIT repository with improper persmissions.  The files are:
```
~/nos3$ ls -la scripts/gsw*
-rw-rw-r-- 1 bj bj  457 Mar 13 21:27 scripts/gsw_cosmos_build.sh
-rw-rw-r-- 1 bj bj  699 Mar 13 21:27 scripts/gsw_cosmos_launch.sh
-rw-rw-r-- 1 bj bj 3831 Mar 13 21:27 scripts/gsw_openc3_build.sh
-rw-rw-r-- 1 bj bj  537 Mar 13 21:27 scripts/gsw_openc3_launch.sh
~/nos3$ 
```

When `make config` is performed (via `all` target), the Python script `scripts/configure.py` is executed.  Based on the `gsw_str` value in `cfg/nos3-mission.xml` two files will be copied from `scripts` to `cfg/build`.  These files are then used by `make gsw` to perform the proper build of GSW software.   Because these shell scripts do not have execute permission the error is generated. 

This error is not seen under VirtualBox Vagrant created VMs because the Vagrantfile configuration overrides file permissions changing all files mounted under `github-nos3` to have their permissions changed to 0770 (og+rwx).

All files in the `scripts` directory of the repository should have 0777 permissions (oga+rwx) to avoid this issue.

bjkowalski, 2024-03-14:
Another minor issue is that the README.md file is missing the `make prep` step which is required before a `make` can be performed.

bjkowalski, 2024-03-14:
The script file `scripts/prepare.sh` attempts to perform `git reset --hard` (line 22) when it is **NOT** inside a git branch.  This generates an error during `make prep`.

Since the purpose of the `reset` is to wipe out local changes to the git clone, and since this follows a `git clone`, I question the value of this line in the script.  At a minimum a line with `cd $USER_NOS3_DIR/cosmos` should be inserted before the `git reset --force` line if it is left in the script.

SpaceCyberCoffee, 2024-04-29:
I am having the following new error when I am trying to install the latest `dev` locally:

[image]

This error also causes the inability to create "Flight software" , "Simulators" and "NOS Time Driver" containers:

```
sc_1  - Flight Software...
bash: cd: /home/user/nos3_local/nos3/fsw/build/exe/cpu1: No such file or directory

sc_1  - Simulators...
Error response from daemon: No such container: nos_terminal
Error response from daemon: No such container: nos_udp_terminal

NOS Time Driver...
Error response from daemon: No such container: nos_time_driver
```

bjkowalski, 2024-05-03:
The latest dev branch throws the following error when performing "make all"
```
make gsw
make[1]: Entering directory '/home/bj/nos3'
./scripts/docker_build_cryptolib.sh
make[1]: ./scripts/docker_build_cryptolib.sh: Permission denied
make[1]: *** [Makefile:95: gsw] Error 127
make[1]: Leaving directory '/home/bj/nos3'
make: *** [Makefile:47: all] Error 2
```

This error is caused by the lack of execution privs on the file "scripts/docker_build_cryptolib.sh".

Performing "chmod oga+x scripts/docker_build_cryptolib.sh" in the "nos3" directory corrects the issue.

closed 2024-05-08 by jlucas9