nasa/nos3 #274: 'NOS3 Ground Systems' Wiki Page Suggestions

bjkowalski, 2024-03-20:
**COSMOS 5 to cFS Command and Telemetry Link Up**

Suggest adding ",as shown in the screenshot below" to the sentence "The Radio TO link can be started using the target named ‘CFS_RADIO’ with a single command to TO_ENABLE_OUTPUT_CC’."

Suggest removing "This is demonstrated in the screenshot below."

After the screenshot suggest adding:

> Telemetry from the Radio TO link can be confirmed by observing that the 'Tx Bytes' and 'Rx Bytes' values are incrementing for the 'RADIO' line in the COSMOS CmdTlmServer window.  Additional ways to confirm successful processing of the command can be accomplished by examinging the output on the 'NOS3 Flight Software' terminal window or the output on the 'Radio Sim' terminal window.  Both windows will have frequent output when the Radio TO link is enabled.

**Toggling Ground System**

Suggest changing the 'Toggling Ground System' heading to 'Selecting Ground System'.  Toggling implies a dynamic nature to the action whereas 'Selecting' implies that additional steps might be needed.  Suggested changes to the following paragraph are included below:

> The Ground System is selected by setting the `gsw` parameter in the `nos3-mission.xml` file in the `cfg` folder. Setting the value to `cosmos` will cause the system to use COSMOS4 as the Ground System.  Changing it to `openc3` will cause the system to use the OpenC3 COSMOS 5. 

>The screenshot below shows the default, COSMOS 4 configuration. Note that in order to switch, if you've run NOS3 previous on you VM, you need to run `make stop-gsw` to ensure that the GSW is stopped. Then you must run `make clean`, `make prep`, and `make`.   After the final make completes the GSW should be operational.  Use the Docker command `docker ps` to verify that the GSW is operational.

bjkowalski, 2024-03-21:
Note that as of 21-Mar-2024 there is no `TO_ENABLE_OUTPUT_CC` packet available to send to target `CFS_RADIO`.  There is only a `TO_ENABLE_OUTPUT` available.

kevincbruce, 2024-03-25:
Hello BJ, thanks for your suggestions! I implemented most of your suggested edits, and fixed some other grammatical issues I found in the original. As for your second addendum, I have changed the name to `TO_ENABLE_OUTPUT`, as that is what displays in COSMOS. I believe the `_CC` addendum is either a name that changed, or what is used for the command code label in C.

kevincbruce, 2024-03-25:
Made all requested edits, and thus closing the issue.

closed 2024-03-25 by kevincbruce