nasa/nos3 #279: "make prep" for Igniter missing dependencies

bjkowalski, 2024-04-16:
"scripts/prepare.sh" within the 234-nos3-igniter branch has been modified to include steps to build Igniter.

Missing from the "pip3 install" on line 44 is "xmltodict".  The line should be:

```
pip3 install pyside6 xmltodict
```

bjkowalski, 2024-04-16:
Consider creating a seperate script called "igniter_launch.sh" with the following contents:

```
#!/bin/bash -i
#
# Convenience script for NOS3 development

SCRIPT_DIR=$( cd -- "$( dirname -- "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )
source $SCRIPT_DIR/env.sh
echo ""
echo ""

cd $BASE_DIR
python3 $BASE_DIR/cfg/gui/cfg_gui_main.py &
echo ""
echo ""
```

Then add a new target to the top level "Makefile" called "igniter" so that a "make igniter" just launches it without performing the other prepare steps.

```
igniter:
    ./scripts/igniter_launch.sh

```

Donnie-Ice, 2024-04-18:
Thanks for that catch!

closed 2024-04-18 by Donnie-Ice