nasa/nos3 #298: OnAIR integration effort

Isgaroth-the-Green, 2024-05-06:
Integrate [OnAIR](https://github.com/nasa/OnAIR) into NOS3 by:

- [ ] add OnAIR to NOS3 in such a way that the two can be compiled and launched simultaneously
- [ ] Connect cFS in NOS3 to OnAIR
- [ ] Confirm/test that communications from cFS to OnAIR are working

Possible additional efforts involve sending messages back from OnAIR to cFS, or such; the three above represent the most basic start to this effort.

kevincbruce, 2024-06-10:
OnAIR now builds properly. Need to assure SBN links are working properly, figure out how to spawn terminals for SBN to monitor traffic, add any necessary commands for SBN or OnAIR to COSMOS dictionaries, etc for proper integration/control.

kevincbruce, 2024-07-19:
With the most recent pushes, OnAIR stays open, and says the Simulation has started. Does not appear further data is going across the bus, but I am not sure if there should be. Seems a good time to check in with James Marshall and his team to move on to step 3 of integration.

jlucas9, 2024-11-15:
Merged into `dev`!

closed 2024-11-15 by jlucas9