nasa/nos3 #305: EPS -  battery voltage behavior

Simon9050, 2024-05-15:
Hi, 

I am using the dev branch of the EPS component and I am surprised by the following behavior of the battery voltage when passing to the eclipse phase :  

[image]

is it normal ? 

moreover, if I select one of the switch to be ON, I cannot see any impact on the battery voltage... what am I missing ? 

Thank you for your help, 

Best Regards,  

Simon

Isgaroth-the-Green, 2024-05-20:
Simon,

regarding the voltage during the eclipse phase, I would expect to see it drop (though probably not quite that precipitously); how long is the satellite eclipsed in your simulation? That is, is the voltage dropping for all of the time the satellite is eclipsed, or only for a portion of that time?

As for the switches impacting the voltage, are you saying there was no instantaneous change, or that the change over time of the voltage was not impacted by turning on one of the switches? I would expect there to be no instantaneous change with the way the voltage and switches are being modeled, but if you are saying that there was no impact on the voltage over time, that would be more surprising. Also, which switch was it?

Simon9050, 2024-05-21:
Hi! Thank you for your reply, 

I think the problem is not related to the eclipse phase as I have the same behavior during a standard simulation (and battery voltage is oscillating every 4 min...). I think there is an issue with my code, but I was not able to find the origin of this strange behavior. 
For the switch, I have tried the number 1 and I was not able to see a clear difference over time on the voltage. 
Could you please share some plot of EPS normal behavior ? 
Thank you again for your support. 
Simon

[image]
[image]

Isgaroth-the-Green, 2024-05-22:
Simon,

when I run the current dev branch, I get a very slowly decreasing battery voltage but no oscillations the way you are describing (I ran it for twenty minutes to be sure). What branch are you running? My first thought is that you are getting overflow errors because the battery voltage is being set wrong; its value is contained in a uint16_t, which goes from 0 to 65535, and measures mV (so that would correspond to where the change is happening). Also, what is the output of the NOS3 FSW terminal (the one which launches in a separate window) when you try to command a change to one of the switches?

Simon9050, 2024-05-29:
Hi! and thank you again for your support 

I am running the main branch and I have added the dev branch modifications only for the EPS component as I was interested in simulating the battery draining. 
Your thought is correct, I have overflow errors on the battery voltage... I was not able to debug my code for now, but I think the problem is related to the sun_vector computation, as I can see that it is not a unit vector and I have very high values on the p_in...  are there specific modifications for 42 code (or other interacting components) that I am missing ? 

For the switches I have the following messages on the FSW terminal that I suppose are ok : 
[image]

Isgaroth-the-Green, 2024-05-29:
Simon,

of course - glad to be of assistance.

I presume that you are not running the most version of main; we pushed a new release to main pretty recently. What is the order of magnitude of the p_in values? I would expect them to be between 0 and 20; it sounds like they are substantially different from that for you. Do you need to be running the older version of main? And if not, could you try with the newer code and see if that solves the problem?

Simon9050, 2024-05-30:
Hi, 
I have tested the last version of NOS3 and I can confirm that EPS values are plausible. I will try to use the new version. 
Thanks,
Have a nice day!

closed 2024-06-14 by jlucas9