nasa/nos3 #306: 42 crashes during extended simulation times

IKuraudo, 2024-05-21:
Hello all,

I am running the newest dev version and noticed that 42 routinely crashes after around 40000-43000 s of simulation time. I noticed the same behavior in an old dev version from last year that I used previously. When keeping the terminal open, the simulator shows the exit code 137. 

[image]

As I want to run the simulation for extended periods of time, I want to ask if there is an artificial limit to the simulation time in 42 that I can set (which I have not found thus far) or if this behavior is an actual issue with 42.

Thank you for your continued support!

Best

kevincbruce, 2024-05-23:
Hello! I have run NOS3 with scripts for extended periods (sometimes 6-8hrs), but I don't think I've run for 11+ before. Will try to put together something to test over the course of around 12 hours and see if I can replicate. Do you know from your previous experimentation if the same effects happen if the time is artificially increased to around that margin?

Thanks for using NOS3, and I'll get back soon!

IKuraudo, 2024-05-23:
Hello Kevin,

Thank you for the reply. I currently do not know where to artificially change the time. Can you guide me to the location?

In any case, I ran simulations for 7+ days at a time when I noticed this issue. Because of the nature of some subsystems requiring a connection to 42, these crashed too when 42 crashed (which is why I can pinpoint the 40000-43000 s of sim_time).

Thanks again for your effort and continued support.
Best

kevincbruce, 2024-05-23:
Hello,

Thank you for the speedy reply! As for artificially setting the time, one can modify the start time in the master config. On the newest dev branch, this will show up in igniter upon make prep. I believe the path to the file can also be found on the Igniter section of the Wiki. The other way would be using the CFE_TIME_SET_TIME command within the CFS_RADIO folder in COSMOS. Though I'm not sure if either of these would really effect the time driver's count on time, which is more likely the cause, as I have experimented with setting the CCSDS time to quite high values artificially and it doesn't seem to crash 42 from my experience.

If you'd like to try changing the start time to around that margin and see if the same thing occurs, feel free. I'll formulate a ~12 hour test script, since that should be over the 43,000 second mark, and see if I have the same issue on a fresh pull of dev.

Thanks again!
Best

kevincbruce, 2024-05-23:
Hello, I got another lead from a coworker - in the file [nos3-base]/cfg/InOut/Inp_Sim.txt, there is a simulation time set. By default it is 86400.0 seconds with a timestep of 0.01 seconds. It seems that is higher than where you start having issues, but increasing the value may help, as when it hits that time, the simulation will shut down automatically.

IKuraudo, 2024-05-23:
Ah, yes, I remember seeing that setting earlier. I always kept that the same, but I can give that a try and report back. 

On another note, sometimes some sims connected to 42 did not crash completely when 42 crashed. I hypothesized that the crashing of the subsystem is due to the subsystem reading an empty/non-existent data point when calling the get_component_data() function. This has generally been an issue when pulling data during a runtime run() function in a hardware model before the full start of 42 (for me, the problem was fixed by waiting for 60 s before the main while loop in the run()). However, I noticed that sometimes, selected subsystems still report data from 42 after the crash, but the whole data point from 42 is filled with all zeros. If you are interested, I can provide you with more data on that tomorrow.

kevincbruce, 2024-05-27:
Hello, I hope you had a good weekend. More data could be useful, yeah. I'll also still try and experiment of my own and see if it crashes after that length of time or not.

IKuraudo, 2024-05-27:
These are two plots from the IMU which is not affected by the shutdown of 42 in the sense that it does not crash. However, you see how all data is all zeros from the rough time I gave you earlier.

[image]
[image]

kevincbruce, 2024-05-28:
So, I don't have as good plots as you at the moment, but running a script and the sim for 12+ hours, I found that the 42 Truth Sim terminal was still sending some sort of output, while the 42 window says saying the time limit was exceeded, and the GUI windows were still in place. So, it seems the best lead is the time limit mentioned before. Have you tried increasing it to see if that helps? If it still doesn't work, I can check other time limits or avenues as well.

kevincbruce, 2024-06-06:
Have confirmed with John, the issue is likely the aforementioned [nos3-base]/cfg/InOut/Inp_Sim.txt Simulation Time parameter. Try increasing it and it should work. If you have any further problems after that step, feel free to contact us again.

closed 2024-06-06 by kevincbruce