nasa/nos3 #307: Table load sequence

girardja, 2024-05-23:
Hi,

I would like to understand how to 
>  modify a table (let say an RTS for example) 
>  compile it and use elf2cbetbl tool to translate it in a tbl file  (or another way if any)
>  upload this tbl file on-board
>  make the tbl available to the FSW
Is there somewhere an example with the commands to use ?

I have the same question for a shared library (.so). I would like to
> modify (let say generic_FSS.so)
> upload it onboard 
> restart the FSS application with the new library _(via CFE_ES_RELOAD_APP)_ 

Thank you very much
Best regards

kevincbruce, 2024-05-23:
Hello!

Thanks for using NOS3! I have not tried doing this for .so files yet, but I have done the process several times for RTS tables. This is what I have done, there may be other viable ways, but this definitely works.

First, copy an existing RTS table's c file (the one you want, or a different one that runs some commands if you want to modify it, just make sure all the places where it says 'rts00X' match the one you are trying to modify. Then, once you have that modified correctly, place it in the [base_dir]/cfg/nos3_defs/tables folder. Then, make clean and make to build NOS3. If your table is in this file, it should build and convert, and automatically be sent to the spacecraft's storage on launch, and will automatically load, validate, and enable. This is all you need to do if you want to have it on the satellite at launch to trigger from COSMOS.

If you want to upload it and overwrite an existing RTS table, then you should look at the file transfer test scripts. These can be modified with the correct paths to delete the existing table with FM, then uplink your RTS table onto the craft. This needs to already be in .tbl form. How I've gotten that is just copying the file from the spacecraft directory while it is running, after verifying it works using the former process, and copying it to my base directory. I then use the launch script to copy them to the temp file where files to uplink are. You can simply copy one of the lines for copying the files for the CFDP tests and just change the file name. However, if you do this, you will then need to manually run the following commands:

`cmd("CFS_RADIO CFE_TBL_LOAD with LOADFILENAME '/cf/sc_rts00X.tbl'")

sleep(CFS_CMD_SLEEP)

cmd("CFS_RADIO CFE_TBL_VALIDATE with ACTIVETBLFLAG 0, TABLENAME 'SC.RTS_TBL00X'")

sleep(CFS_CMD_SLEEP)

cmd("CFS_RADIO CFE_TBL_ACTIVATE with TABLENAME 'SC.RTS_TBL00X'")

sleep(CFS_CMD_SLEEP)`

These will get the table loaded and working properly in flight software.

I am going to try and update the wiki to contain a more clear tutorial on all the steps for this, with pictures, code, examples, etc within the next week or so, so check back if you need more details or information!

girardja, 2024-05-27:
Hi Kevin, thank you very much for your answer.

When you say for example
cmd("CFS_RADIO CFE_TBL_LOAD with LOADFILENAME '/cf/sc_rts00X.tbl'")
I guess it is a pseudo code to indicate that we have to send manually (from COSMOS) the command CFE_TBL_LOAD with the right arguments ?
Or is it this a real line to put somewhere in a script (where ?) 

More generally, is there a mean to script commands and launch them like batch files ?
Might ATS be a way to do that ?

kevincbruce, 2024-05-27:
Hello! So, COSMOS (both 4 and 5) can be scripted through Ruby. Scripts go in [base-nos3]\gsw\cosmos\config\targets\MISSION\procedures. If you add a new folder/test suite, you need to add it into mission_test.rb in a couple places to match the others. If you add more scripts, you need to add them in a similar way, either triggering the script with _start([file_path_from_here]) or you can put the literal script in mission_test.rb. Commands are run in the format I have there. the sleeps and the like are important to intersperse at least every few commands to ensure things don't get spammed to a point where they do not run correctly. These scripts can then be run from test runner, or copied into script runner, and will control the spacecraft automatically. The other key for this particular operation is you should do is make sure radio telemetry is active with: cmd("CFS_RADIO TO_ENABLE_OUTPUT with DEST_IP 'radio_sim', DEST_PORT 5011") before doing anything else in the script. There should be some others in there that give examples of what to do. Otherwise, if you need more, ask and I can find some good examples, and maybe provide a tutorial here on basic COSMOS mission scripting in Ruby.

kevincbruce, 2024-05-27:
Also, the file transfer scripts exist as literals within the mission_test.rb file as well, if you are still looking for them.

girardja, 2024-05-27:
**Thank you so much, Kevin**. I have not explored the ruby files yet. :-(
I think that any "good examples" and tutorial are worth sharing with the community because there are plenty of implicit behaviors that are not obvious to understand at first glance.

kevincbruce, 2024-06-06:
I hope this guide was good enough for now. There is a new issue for Documentation on mission setup (#314) if you have any feedback on guide that would be useful. I've started adding some information on tables to that, and will add file uploads fairly soon. I'm going to close this issue for now, and feel free to bring any ideas and feedback to the discussion on #314 as guides are developed!

closed 2024-06-06 by kevincbruce

gardinwin, 2024-12-11:
You can make table image with code.
In elf2cbetbl use table heare, and RTS/ATS command data.
You can make table image with your code and load with scripts.