nasa/nos3 #309: Startup and scheduled commands in openc3 with new hardware models

IKuraudo, 2024-05-27:
Hello everyone,

Using the newest dev branch, I have constructed two new subsystems (solar panel and battery) from the sample component and connected them to the nos3 simulation and openc3. Also, I assigned exclusive UART ports in the fsw and the hardware models and exclusive command msg ids in `component/fsw/platform_inc/component_platform_cfg.h` and both files in the `component/gsw/COMPONENT/cmd_tlm/` directory.

A weird thing that I noticed is that openc3 seems to send NOOP commands to the fsw without end, but only to the solar panel, not the battery, and none to other subsystems (edit: the uart port for the solar panel and all existing subsystems are shown as connected to the fsw in the NOS engine server, the one for the battery is not). In addition, the "received NOOP command" shows up in the fsw terminal but not the hardware model terminal, even though the debug print is active in the hardware model. If I enable the solar panel via a command in openc3, I see that it also receives a constant housekeeping command, which is then also shown in the respective sim terminal. 

[image]
[image]

In order to get to the bottom of this problem, I wanted to investigate the command sending within cosmos. Can you point me towards the “scheduler” of the COSMOS command sending that is responsible for the startup command sequence and the command loop that is run during the simulation? In my time working with nos3, I have not located it yet.

Thank you for your assistance and continued support!

Kind regards

IKuraudo, 2024-06-03:
Update: I added the REQ_HK commands to unused cmd IDs in `cfg/nos3_defs/tables/sch_def_msgtbl.c` but not the `sch_def_schtbl.c` in the same directory. Now at least the scheduling works. 
I would still like to know how I can edit the startup commands (e.g., the sample gets an enable command in the beginning, but not the new sim for example)

jlucas9, 2024-06-25:
@IKuraudo - apologies for the delay getting back to you. The latest commit in [nos3#315](https://github.com/nasa/nos3/commit/7c6fc4d0c3cc37f1bdf9506f1943a98fe4551f27) gives an example of how this could be done.

jlucas9, 2024-07-23:
Closing due to age, please re-open or file a follow up issue as needed!

closed 2024-07-23 by jlucas9