nasa/nos3 #311: COSMOS5 GSW problem

Gonta01, 2024-05-30:
Tried to use COMSOS5 but while trying to send commands runtime error occurs.
[image]

jlucas9, 2024-06-19:
Just walking through my steps to attempt to recreate:
* git checkout dev
* git submodule update --init --recursive
* git rev-parse --short HEAD
  * 9b56e75
* make prep
* In Igniter
  * Change GSW to `openc3`
  * Save
* make
  * Wait for build to complete and CPU utilization to settle down as OpenC3 COSMOS continues to process in the background
* make launch
  * Reorganize windows
  * Set OpenC3 cosmos password

Confirmed debug data being received in OpenC3 COSMOS CmdTlmServer:

[image]

Manually send ARDUCAM_DEBUG CAM_NOOP_CC:

[image]

I think it could have been that the CmdTlmServer was not connected when you tested originally. Let me know if you have any follow-ups to this!

Gonta01, 2024-06-20:
I tested with 1.7.0 release and the result is the following:
[image]
[image]

Tried to send command:
[image]

jlucas9, 2024-06-25:
Looks like your interfaces didn't get created successfully in the CmdTlmServer. Can you `make stop-gsw` then do another `make prep` and `make` and see if it builds and attaches the OpenC3 plugin correctly the second round?

jlucas9, 2024-07-23:
Closing due to age, if this problem persists please re-open or file a new issue with your latest!

closed 2024-07-23 by jlucas9