nasa/nos3 #312: COSMOS4 problem when closing windows

Gonta01, 2024-05-31:
When I close any window in COMSOS4 the following window pops up:
[image]

Isgaroth-the-Green, 2024-06-03:
Good morning.

I have experienced the same thing as you, but as far as our team can tell it does not have any deleterious effects. You should be able to simply click "OK" or close the window and continue using NOS3. 

Also, this is the version of COSMOS we typically use; COSMOS5 adds quite a bit of overhead.

jlucas9, 2024-06-25:
@Gonta01, can you try `dos2unix ./generate_template.sh` and see if that resolves it for you?

Gonta01, 2024-06-25:
@jlucas9 I configured VSCode so that end of line sequence is LF not CRLF inside files, so I don't encounter problem neither with COSMOS4 nor with generate_template.sh.

closed 2024-06-25 by Gonta01