nasa/nos3 #313: Is it possible to simulate ADC?

Gonta01, 2024-06-03:
Hi.
Suppose that I want simulate receiving of analog data by ADC(Analog to Digital converter). Assume that this part of calculating temperature with the usage of ntc sensor.
How could it be accomplished in NOS3 framework?
I thought it could be something similar to usage of USART, I2C, CAN but these are communication protocols and (as far as I understand) NOS engine was designed to provide simulation for communication protocols.
I am new to this framework, so some of my statement may be wrong. Any hints are appreciated.

Isgaroth-the-Green, 2024-06-03:
Good morning.

I believe that would probably be best done in the NOS3 framework by creating an ADC component or incorporating an ADC into whichever component it is that you want to receive the analog data. In the first case, you would bring in the analog data either using your own files or through 42, if it is data which 42 simulates, and then your ADC component would convert to a digital format. This would then send the data onward to whichever component needs it (which it sounds like would also be a custom component). In the second case, you would do largely the same as in the first except that the two components would be merged.

Gonta01, 2024-06-04:
@Isgaroth-the-Green thank you for reply. I've read [NOS3 Hardware Simulator Framework, Example Simulator article](https://github.com/nasa/nos3/wiki/NOS3-Hardware-Simulator-Framework,-Example-Simulator) but it isn't clear for me how to use own files or 42 to provide data. Could you please give a hint where it is described or what parts of project I should to watch to  understand this or maybe you could explain it?

Gonta01, 2024-06-04:
Let assume that I have component NTC (temperature_sensor) without ADC component and I want to simulate that some temperature values are provided. What should I do? Could I use 42 somehow or I need to use some files as you mentioned before?

Isgaroth-the-Green, 2024-06-04:
Certainly. If your goal is to use 42 data, you could look at existing components which use such data (such as the generic IMU or the generic FSS and CSS); the data is read into the simulator (generic_imu_42_data_provider, I believe, is one such example of a file) and then passed on to the other pieces of the simulator. The 42 documentation would provide information regarding which temperature values can be provided.

Gonta01, 2024-06-05:
@Isgaroth-the-Green. Okay lets assume that I receive simulation in my NTC module it is then transmitted to flight software. For now, it is transmitted through uart(because I used sample project to create new component) but in reality NTC isn't UART device. Are there any other ways to transmit data to flight software (despite using communication protocols)? I considered [Generic torquer](https://github.com/nasa-itc/generic_torquer/tree/bd9242dceba6c982191efdb9cb8a0aaedf4aad65) as closest example because there are no any communication protocols included but I don't see the way how it transfers data to FSW. It seems to me that this module receives from 42 and then again sends to 42 simulator. In ground system there are no data telemetry of generic torquer, only housekeeping.

Isgaroth-the-Green, 2024-06-05:
The generic torquer does not send information other than housekeeping to the FSW, because there is (for the most part) no information to transmit - the torquers are commanded devices. Regarding other ways to transmit data to flight software, are you looking for something using a different communication protocol (CAN, or I2C, or some such thing)? Or something which uses no communication protocols? And if the latter, what are you envisioning it ought to do?

Gonta01, 2024-06-06:
@Isgaroth-the-Green, this is a device without communication protocol. Voltage value is received by microcontroller from voltage divider (one of the resistors is thermal resistor). Based on this voltage value the temperature is calculated by microcontroller. This module I call NTC.

Gonta01, 2024-06-19:
@Isgaroth-the-Green could you comment somehow my last comment? Maybe this kind of module could not be simulated and that's all?

jlucas9, 2024-06-25:
Hi @Gonta01, unfortunately you'll need to have a communication protocol of some kind in order to communicate with the simulations in NOS3. A typical ADC on a PCB will use I2C or SPI to talk back to the processor or microcontroller. An example of this is the generic_css or coarse sun sensor which essentially is an ADC with multiple channels, one for each CSS around the outside of the spacecraft to detect the sun. 

Let us know if you have any follow up questions and have a great day!

Gonta01, 2024-06-25:
@jlucas9 thank you for explanation!

closed 2024-06-25 by Gonta01