nasa/nos3 #314: Improved cFS/COSMOS Documentation for Mission Creation

kevincbruce, 2024-06-06:
Due to Open Source user feedback, we plan to start creating guides on how to properly develop missions in NOS3, starting with commonly edited tables and COSMOS scripting. So far, a basic guide on tables is added to the Wiki, which should explain the use of DS tables (File and Filter) in depth, with guides on SC RTS tables and SCH tables soon to follow on that same page. We also plan to add a guide on COSMOS Scripting in Ruby, and perhaps other commonly used features. If users, or developers, have any inputs on these guides, either in what we should add, or comments on existing guides for their accuracy, detail, or if they are flight-like, please post them here so we can foster discussion and a comprehensive guide for our users.

Gonta01, 2024-06-07:
Hi. 
The guides that will be appreciated:
1) How to add custom component; So far, the only way is to use [sample](https://github.com/nasa-itc/sample/tree/34260dd8a8d929b6d36c8805f08754aa847d6e97), but steps described there is not very detailed and full:
[image]

2) Probably this is more related to cFS not NOS3, but it will be interesting to get some details on that how NOS3 could be ported for FreeRTOS and real board.

jlucas9, 2025-03-26:
Closing due to age, a number of new issues work to resolve providing more detailed tutorials and scenarios for use.

closed 2025-03-26 by jlucas9