nasa/nos3 #315: Initialization of a new Sim starting from a specific snapshot of the previous one.

lucapizzuto99, 2024-06-11:
Hello everyone,

I am working with NOS3 to try to initialize a new simulation, that is a new spacecraft, given the orbital conditions and the TM of the originally simulated spacecraft at a time t1.

doraemox, 2024-06-14:
hello, nice to meet you all

jlucas9, 2024-06-19:
Hi @lucapizzuto99,

I think you're on the right path here and would need to modify the configuration files (primarily the 42 ones) and then launch everything again to resume. Presently we don't have  a means to save the state of the entire environment and launch it again so it would assume you just turned on, like a full power reset of the entire spacecraft, at that location and time and continue. 

If 42 is launched with the new location and time, the rest of the sims will receive updated values and translate those into there own messages as you'd expect, assuming they're configured to reference environmental data that is.  

Starting with all subsystems enabled - that's a great idea. I've created a new branch `nos3#315` to update the relative time sequence (RTS) commands to do this for you. A brief walkthrough of bootup now:
* Spacecraft turns on
* RTS1 (Power on reset) automatically fires
  * Enables data storage (DS) to save telemetry to files
  * Enables debug port so we can view what's going on in the ground software as if we're plugged into the spacecraft
  * Enable RTS3-64 so that they can be used
  * Enables the limit checker (LC) application to perform fault detection and correction, note that none is presently configured
  * Call RTS3
* RTS 3 (Safe mode) 
  * Enable coarse sun sensors (CSS)
  * Enable fine sun sensors (FSS)
  * Enable inertial measurement unit (IMU)
  * Enable magnetometer (MAG)
  * Enable torquers
  * Enable GPS
  * Set ADS to SUNSAFE_MODE

Let me know if you have any other questions!

lucapizzuto99, 2024-06-25:
Hi @jlucas9,

Thank you so much for your support and quick reply! Starting with all subsystems enabled works as expected. Regarding the initialization of a new sim starting from a previous state, everything works as well, with the exception of the IMU whose linear acceleration values are a bit off, even if the S/C has the same state vector and attitude. I will explore this a bit more.

Thank you again!

jlucas9, 2024-06-26:
Merged into `dev`! Feel free to re-open or create a new issue as you see anything off!

closed 2024-06-26 by jlucas9