nasa/nos3 #321: Custom simulator exits

Gonta01, 2024-06-19:
Created custom simulator (hardware model) but it exits at the very beginning so that I could not even see logs.
Could somebody give debugging tips for this situation?

jlucas9, 2024-06-19:
Hi @Gonta01, can you confirm that your hardware model key is in the configuration file being used?

Looking at the default sample sim output for example:
[image]

The sample simulator contains the same <hardware-model> / <type> key that is listed in the printed configuration it read - https://github.com/nasa-itc/sample/blob/dev/sim/src/sample_hardware_model.cpp#L5

If I change the following:
* sample_hardware_model.cpp line 5
  * `REGISTER_HARDWARE_MODEL(SampleHardwareModel,"SAMPLE_ERROR");`
* docker_launch.sh line 130
  * `    gnome-terminal --window-with-profile=KeepOpen --title=$SC_NUM" - Sample Sim"   -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name $SC_NUM"_sample_sim"   --network=$SC_NETNAME -w $SIM_BIN $DBOX ./nos3-single-simulator $SC_CFG_FILE sample_sim`
  * Assuming the KeepOpen profile exists in gnome-terminal as you [linked](https://github.com/nasa/nos3/issues/260#issuecomment-1971126479)

[image]

Let me know if things still aren't working for you and I can take a further look!

Gonta01, 2024-06-20:
@jlucas9 thank you for your response! In my case names are the same. 
I have looked through logs of Sim more carefully and found the following message:
[image]

Trying to find out why is it happening.

Gonta01, 2024-06-20:
I created my own application STS21 on the basis of Sample and after this tried to use I2C instead of UART and probably some configurations are missed

Gonta01, 2024-06-20:
I've found how to fix the [problem](https://github.com/nasa/nos3/issues/321#issuecomment-2179989758)
Instead of uart I should write i2c and this problem doesn't occur again
[image]

Gonta01, 2024-06-20:
@jlucas9 thank you again.
In my opinion this particular issue is solved.

closed 2024-06-20 by Gonta01