nasa/nos3 #322: COSMOS receives error messages

Gonta01, 2024-06-20:
Added new component and seems to resolve all troubles related to it (including [issue](https://github.com/nasa/nos3/issues/321))
Some problem remain but I'm not sure what could cause it. Just after the start of the system and activating of COSMOS, COSMOS receives the following messages:
[image]

Where could be hidden the problem and how to debug it?

Gonta01, 2024-06-20:
In Tlm packets found the following:
[image]

There is UNKNOWN target and my newly added component(STS21). Probably they are not related.

Gonta01, 2024-06-20:
I has found that Software Bus task houskeeping Packet starts with 0x0803
[image]

Gonta01, 2024-06-20:
Compared CFS and CFS_RADIO targets.
[image]
[image]

It seems that CFE_SB_HKMSG, CFE_SB_PREVSUBMSG, CFE_SB_STATMSG and CFE_SB_SUBPRTMSG in case of CFS target (not CFS_RADIO) for some reason are labeled as UNKNOWN and somehow influence CFE_TBL_HKPACKET based on logs:
[image]

Gonta01, 2024-06-21:
In cmd_tlm_server.txt and system.txt of gsw commented RADIO target of STS21 and problem SEEMS to be fixed:
[image]
[image]

CFE_SB_HKMSG, CFE_SB_PREVSUBMSG, CFE_SB_STATMSG and CFE_SB_SUBPRTMSG appeared in CFS target:
[image]

Are there any limitations of targets for this ground station (COSMOS 4)?

kevincbruce, 2024-06-24:
Hello @Gonta01, I hope you're well! Did you add a COSMOS Dictionary for your app? They'd be found in nos3/gsw/cosmos/COMPONENTS. You'd add a directory with all the equivalent stuff to what's there for other apps for your app, and define your packets in the way found in the other dictionaries. That way, COSMOS recognizes your APP_ID, MIDs, CCs, etc. correctly, and knows what lengths to look for and what values to put where. Should also make it show up in the packet viewer and  command sender and the like. If you've already done it, is there a misconfiguration/misplaced file, or a formatting error within one of those files? Hope that helps!

Gonta01, 2024-06-25:
Hi @kevincbruce. I have uncommented [STS21_RADIO components](https://github.com/nasa/nos3/issues/322#issuecomment-2182123621), built again and it seems to be everything good for now.
Have no idea what was the root of the problem and how it was fixed.

jlucas9, 2024-06-25:
@Gonta01 this could have been related to MSGIDs clashing. It appears that CFE_SB_HK_TLM_MID is also 0x0803 - https://github.com/nasa/nos3/blob/dev/cfg/nos3_defs/cpu1_msgids.h#L118

Gonta01, 2024-06-25:
Newly added component had and has unique MSGIDs:
STS21_CMD_MID              0x18A0
STS21_REQ_HK_MID           0x18A1

Nevertheless, for now issue was somehow fixed (by disabling new component, and further enabling and building?). In case, if this problem appears again, I'll let you know and reopen this issue.

closed 2024-06-26 by jlucas9