nasa/nos3 #329: Git Cleanup

jlucas9, 2024-07-09:
`git fsck` on a fresh clone shows the following errors:
```
$ git fsck
Checking object directories: 100% (256/256), done.
Checking objects: 100% (13943/13943), done.
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/deployment.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/CFS_CI.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/CFS_TO.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/CFS_IO_LIB.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/cFE.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/osal.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/PSP.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/cFS-GroundSystem.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/elf2cfetbl.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/SCH.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/SC.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/CF.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/HK.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/cfs_lib.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/LC.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/hwlib.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/HS.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/arducam.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/clyde_eps.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/novatel_oem615.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/clyde_battery_sim.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/arducam_sim.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/clyde_eps_sim.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/nos_time_driver.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/novatel_oem615_sim.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/sim_common.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/sim_server.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/sim_terminal.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/gsw-ait.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/OrbitInviewPowerPrediction.git
error in blob 01e1d1815038cf1f6d519c6d039d272c5d249731: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/gsw-cosmos.git
error in blob 525de1a5c5b0198aa3c510f5747161e54b74d002: gitmodulesUrl: disallowed submodule url: https://github.com:nasa-itc/generic_css.git
Verifying commits in commit graph: 100% (668/668), done.
```

Need to resolve this as this stops GitLab from doing a mirror of the repository presently.

jlucas9, 2024-07-15:
May be related to https://github.com/actions/runner-images/issues/9883

jlucas9, 2025-02-12:
@dgeletko - Any chance you can help us out with this one? I think we may need to clean up some history to resolve it.

jechiasson, 2025-04-29:
Here's the final commands I used:
```
git filter-repo --replace-text <(echo '.gov:==>.gov/') --force --refs dev
git filter-repo --replace-text <(echo '.com:==>.com/') --force --refs dev
git filter-repo --replace-text <(echo 'git@:==>https://') --force --refs dev
```

I'm not sure which of the ref rewriting options to use: [docs](https://www.mankier.com/1/git-filter-repo#Options-Parent_rewriting)

jechiasson, 2025-05-19:
@jlucas9 I finally have a consistent way to change .gitmodules commits to https style urls.  This will rewrite all the refs in the repo and is recommended to be run against a fresh pull.  It will also remove your remote url as a safety measure so it will need to be re-added to force push the rewritten history.

Find all .gitmodules non https style urls
`
git rev-list --all | while read rev; do
  git grep -E '\.gov:|\.com:|git@' $rev -- .gitmodules
done
`

Re-write .gitmodules commits using https style urls: 
`
git-filter-repo --file-info-callback '
import re
if filename == b".gitmodules":
    contents = value.get_contents_by_identifier(blob_id)
    content_str = contents.decode("utf-8")
    content_str = content_str.replace(".gov:", ".gov/")
    content_str = content_str.replace(".com:", ".com/")
    content_str = content_str.replace("git@", "https://")
    new_contents = content_str.encode("utf-8")
    new_blob_id = value.insert_file_with_contents(new_contents)
    return (filename, mode, new_blob_id)
else:
    return (filename, mode, blob_id)
' --force
`

If your team could do this for nos3 and the sub repos it would really ease mirroring and updates for gitlab users.