nasa/nos3 #330: Help with routine operation

Simon9050, 2024-07-09:
Hi,

I am using NOS3 and I would like to create a STF-1 (or similar based on NOS3) routine operation scenario. 
Could you provide a representative list of TCs that I can use in Cosmos (and that represents for example a few days of routine operations) ? Or could you suggest someone that can provide this type of info ?

Thank you very much for your help,
Best Regards,
Simon

Simon9050, 2024-07-23:
Hi @jlucas9,

thank you for your feedback and for your support, 

May I ask you what type of scenario do you use for testing NOS3 ?  I can probably start from something very simple and then improve ... 

For example, do you think that starting from a short list of TC as https://github.com/nasa-itc/gsw-cosmos/blob/50c889a4f6ad21bd0dbc6e32254252a435e57770/procedures/deployment_test.rb

Could be a good starting point ? 

Could you suggest some typical TC for health checking, to activate correctly the radio or other typical components, to do some data transfer, etc ...

I think that a few examples could be really helpful to better understand the simulator behavior.  

Thanks and have a great day too! 

Simon

jlucas9, 2025-05-06:
Merged into `dev` ^

PR #611 (2025-04-08): [nasa/nos3#330] Help with routine operation
To test:
Review `docs/wiki/Nominal_Operations.md` to make sure it makes sense, is complete, is understandable, etc.
Also review `docs/wiki/index.rst` to ensure that the `Nominal_Operations.md` page seems to be in the correct location in the index.

Closes #330
closed 2025-05-06 by jlucas9