nasa/nos3 #332: Question regarding external application

matt11matthew, 2024-07-12:
Would it be possible to listen on the ports with an external application and send/receive packets from NOS3? I'm hoping to create an external application frontend that allows for more simple vector manipulation. I just am looking for some advice as where I should start.

matt11matthew, 2024-07-12:
Also my hope is to develop my frontend in Java as well.

kevincbruce, 2024-07-12:
Hello Matt, I hope you're doing well! It is possible to listen to ports with external applications. I have done so before on multiple occasions with wireshark, though that picks up the CCSDS messages (the protocol used for space), and you'd still need to write a parser to do much with them. Especially some of the things contained in the data section of the message may be formatted differently or mean different things depending on what packet you are looking at, and you would need to have some sort of dictionary like we have for COSMOS to decode it. 

So, for the backend, you'd need to listen to the proper ports on the network at the least (and you may want to filter to specific docker container IPs as well if you are running multiple spacecraft), then parse the packets to get what information you need, then send it to your frontend for display.

I hope that helps, let me know if there are other things you need for support!

matt11matthew, 2024-07-12:
Thank you, are you able to point me in the right direction regarding references? I'm having trouble figuring out what packets I'm supposed to use for manipulating vectors. I'm very new to this type of research so any guidance would be appreciated.

kevincbruce, 2024-07-12:
By manipulating vectors, do you mean controlling the Attitude? If so, you should look at the Reaction Wheel, Torquer, and ADCS components and their packets. If you want specific references, you can look through them in COSMOS, the COSMOS dictionaries in the file system, or at their header files. If you want references on CCSDS protocol standers, I could send some as well. It sounds like you want to create your own custom ground software (correct me if I am wrong), so the COSMOS dictionaries are probably a good starting place to see how it builds commanding packets.

matt11matthew, 2024-07-12:
That would help a lot, I'm currently figuring out what ports to listen to, I managed to get some output from the debug port 9999 using Wireshark like you said. The output seemed to come out as encrypted or just in binary. So my next step would be to figure out how to decrypt using CrypticLib and then use the protocol standards to some how parse the binary into a readable format. My hope is to be able to use that to create a spring boot application that can communicate with the front end.

kevincbruce, 2024-07-12:
If you listen on the right ports, you can get it before or after it is encrypted. I believe 6010/6011 are the ports that wouldn't be for the radio, if you enable it. I forget what it is for debug, though I can check quickly.

matt11matthew, 2024-07-15:
Would there be any documentation with all of the ports and packets?

kevincbruce, 2024-07-16:
Sadly, I'm not sure how in depth it goes. If you look around the code for ports, you can generally find a lot of them. https://github.com/nasa/nos3/blob/dev/cfg/sims/nos3-simulator.xml Has a decent number of them listed. This wiki page has ranges of message IDs for various apps and components, though to find the specific ones you need to got to each app/component's MID file: https://github.com/nasa/nos3/wiki/NOS3-Component-Directory.

closed 2024-07-18 by matt11matthew