nasa/nos3 #334: Reading .txt files within NOS3

lucapizzuto99, 2024-07-16:
Hi everyone,

I am trying to open and read contents of .txt and .csv files from the fsw of a specific component, just like the generic_adcs does when reading the input quaternions for the subsystems, for example. 
I am using fopen to open the .txt file and here is the simple code I am using:

```
FILE *file_in = fopen("Inp.txt", "r");
 if (file_in == NULL) {
           CFE_EVS_SendEvent(SAMPLE_PROCESS_CMD_ERR_EID, CFE_EVS_EventType_ERROR, "Error opening Inp_DI.txt");
  }
 Readthisfile(file_in);
 fclose(file_in);
```

Despite the file is in the correct location and the user jstar is part of the vboxsf group, which has read and write permissions, fopen returns NULL and the file is not read by the next method. 

Any idea on how I can solve this issue? I would higly appreciate your guidance.

Thanks in advance!

Cheers,
Luca

szemerick, 2024-07-16:
The code you are showing is running from a cFE context, so the fopen() function is going to be looking for "Inp.txt" in a location that is relative to the cFE startup folder.  Is this where "Inp.txt" is actually located? Try using absolute file paths and see if you get any different results.  Also, you can check where you current working directory [getcwd()] is just as a sanity check.

lucapizzuto99, 2024-07-17:
Hello @szemerick , 
Thank you for the hint, indeed I did not consider that my function was actually running from the cFE startup folder, and that was exactly the issue. With the absolute path it works without problems. However, the issue perists in the hardware model. Despite the absolute path, the file is not found and not opened.

`std::string fileName = "/home/jstar/Desktop/github-nos3/components/sample/sim/src/Inp.txt"; 
        char cwd[120]; 
        if (getcwd(cwd, sizeof(cwd)) != nullptr) {
            sim_logger->debug("Current working directory: %s",cwd); // /home/jstar/Desktop/github-nos3/sims/build/bin
        } else {
            sim_logger->debug("Current working director error");
        }

        std::ifstream file(fileName);
        if (!file) {
            sim_logger->debug("Error opening file:");
        }

        std::string line;
        while (std::getline(file, line)) {
            sim_logger->debug("toh file %s", line.c_str());
        }

        file.close();
`

lucapizzuto99, 2024-07-18:
Update:

I found out that a file, given its absolute path, can be opened only if it is saved in the `/home/jstar/Desktop/github-nos3/sims/build/bin `directory, or in every subdirectory created within bin, which is the directory where the subsystem simulation is running.

jlucas9, 2024-07-23:
Hi @lucapizzuto99,
It looks like this is resolved. If you have follow up questions please feel free to re-open or create a new issue as you see fit.

closed 2024-07-23 by jlucas9