nasa/nos3 #337: Cannot install on Win11

wensijie, 2024-07-18:
Hello,
I have tried to install NOS3 in Win11, following the instructions.
System：Windows 11 23H2;       Vagrant: 2.4.1           VirtualBox: 7.0.20

After running "vagrant up", an error message appears:
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Box 'nos3/ubuntu' could not be found. Attempting to find and install...
    default: Box Provider: virtualbox
    default: Box Version: 20231101
==> default: Loading metadata for box 'nos3/ubuntu'
    default: URL: https://vagrantcloud.com/api/v2/vagrant/nos3/ubuntu
The box you're attempting to add has no available version that
matches the constraints you requested. Please double-check your
settings. Also verify that if you specified version constraints,
that the provider you wish to use is available for these constraints.

Box: nos3/ubuntu
Address: https://vagrantcloud.com/api/v2/vagrant/nos3/ubuntu
Constraints: 20231101
Available versions: 0.0.0, 1.6.0, 1.6.1, 20230630, 20230714, 20230718, 20230731, 20231101

Do you know what's wrong?
Thanks,

jlucas9, 2024-07-23:
Hi @wensijie,
This appears to be an off and on problem with vagrant for some reason that we're still trying to figure out a permanent work around to. For now you can either create your own Linux VM or simply download this OVA directly:
* https://drive.google.com/file/d/1ZcYLkPDy8aNgwTfQ9hUDvTP40XcRx1L9/view?usp=sharing
Note that this OVA does not have shared folders set up so the path to `~/desktop/nos3` in the readme is invalid. Instead this OVA has NOS3 pre-installed in `~/Downloads/nos3`.

jlucas9, 2024-08-30:
Closing due to age. If you still have problems please re-open or create a new issue!

closed 2024-08-30 by jlucas9

ephraim271, 2025-02-11:
@jlucas9 Still have this issue on Ubuntu 22.04, do you know a fix for this?

ephraim271, 2025-02-11:
@jlucas9 Update : I had the wrong version of virtual box and vagrant pre-installed. After updating to the ones in the readme, it works now!