nasa/nos3 #338: Installing directly on Linux

matt11matthew, 2024-07-18:
Is there a certain process I would need to follow for installing NOS3 directly on a Linux machine? I would just clone the repo with docker installed and run make clean, make prep, make, etc...? Just confirming that there aren't anymore additional steps before requesting a independent Linux machine since the machine I'm using for a VM is not powerful enough to run Cosmos5.

gardinwin, 2024-07-23:
I have received a mail from NASA :
-------------------------------------------------------------------------------------------------------
For option A you can simply clone the repository then following the instructions under Getting started (https://github.com/nasa/nos3?tab=readme-ov-file#getting-started) assuming you navigate into the cloned repository. For example:

 

    git clone https://github.com/nasa/nos3.git
    cd nos3
    git checkout dev
    git submodule update --init --recursive
    make prep
    make
    make launch

-------------------------------------------------------------------------------------------------------------------------------------------------------------
but make step has error :
sudo make
make config
make[1]: Entering directory '/home/polemotion/nos3'
./scripts/config.sh
  start-time: 814048200.0
  start-time-utc: 2025-10-18 08:30:00
  gsw : cosmos
  number-spacecraft: 1
make[1]: Leaving directory '/home/polemotion/nos3'
make fsw
make[1]: Entering directory '/home/polemotion/nos3'
./scripts/docker_build_fsw.sh
mkdir -p /home/polemotion/nos3/fsw/build
cd /home/polemotion/nos3/fsw/build && cmake -DCMAKE_INSTALL_PREFIX=exe -DCMAKE_BUILD_TYPE=debug ../cfe
CMake Error: The current CMakeCache.txt directory /home/polemotion/nos3/fsw/build/CMakeCache.txt is different than the directory /home/jstar/Desktop/github-nos3/fsw/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/polemotion/nos3/fsw/cfe/CMakeLists.txt" does not match the source "/home/jstar/Desktop/github-nos3/fsw/cfe/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
make: *** [Makefile:56: build-fsw] Error 1
Makefile:92: recipe for target 'fsw' failed
make[1]: *** [fsw] Error 2
make[1]: Leaving directory '/home/polemotion/nos3'
Makefile:44: recipe for target 'all' failed
make: *** [all] Error 2

gardinwin, 2024-07-23:
i success install after clean optionB
but python cannot launch gui error, i will upgrade ubuntu or python version.

jlucas9, 2024-07-23:
Let us know if you continue to run into issues you think we can help with!

matt11matthew, 2024-07-23:
Would option A still work since I noticed someone else had issues with it? I don't have a way of creating a high powered VM so using option A directly on a Linux machine would be my only option for getting high GPU performance

gardinwin, 2024-07-24:
Option A, make prep, make passed success, upgrade python version with mini conda.
Depending on the version of Ubuntu, an upgrade of the Python version is required. When running `make launch`, there is a "docker container not found" error and the UI like 42 does not appear due to OpenGL issues.
Anyone success in launch step ?

gardinwin, 2024-07-24:
aftere version upgrade (ubuntu 18 -> 20), almost service are running ; but dbus error occured.
which version is good for nos3 ?

matt11matthew, 2024-07-28:
Can someone help with the docker issue?

matt11matthew, 2024-07-29:
The issue has been resolved.

closed 2024-07-29 by matt11matthew

gardinwin, 2024-08-01:
I solved problems.
in Makefile, i added sleep n seconds before start new docker.
I think before docker exec. next docker connection fails.
Have a good day.

gardinwin, 2024-09-04:
#!/bin/bash -i
#
# Convenience script for NOS3 development
# Use with the Dockerfile in the deployment repository
# https://github.com/nasa-itc/deployment
#

SCRIPT_DIR=$( cd -- "$( dirname -- "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )
source $SCRIPT_DIR/env.sh

# Check that local NOS3 directory exists
if [ ! -d $USER_NOS3_DIR ]; then
    echo ""
    echo "    Need to run make prep first!"
    echo ""
    exit 1
fi

# Check that configure build directory exists
if [ ! -d $BASE_DIR/cfg/build ]; then
    echo ""
    echo "    Need to run make config first!"
    echo ""
    exit 1
fi

echo "Make data folders..."
# FSW Side
mkdir $FSW_DIR/data 2> /dev/null
mkdir $FSW_DIR/data/cam 2> /dev/null
mkdir $FSW_DIR/data/eo_cam 2> /dev/null
mkdir $FSW_DIR/data/evs 2> /dev/null
mkdir $FSW_DIR/data/hk 2> /dev/null
mkdir $FSW_DIR/data/inst 2> /dev/null
# GSW Side
mkdir /tmp/nos3 2> /dev/null
mkdir /tmp/nos3/data 2> /dev/null
mkdir /tmp/nos3/data/cam 2> /dev/null
mkdir /tmp/nos3/data/eo_cam 2> /dev/null
mkdir /tmp/nos3/data/evs 2> /dev/null
mkdir /tmp/nos3/data/hk 2> /dev/null
mkdir /tmp/nos3/data/inst 2> /dev/null
mkdir /tmp/nos3/uplink 2> /dev/null
cp $BASE_DIR/fsw/build/exe/cpu1/cf/cfe_es_startup.scr /tmp/nos3/uplink/tmp0.so 2> /dev/null
cp $BASE_DIR/fsw/build/exe/cpu1/cf/sample.so /tmp/nos3/uplink/tmp1.so 2> /dev/null

echo "Create ground networks..."
$DNETWORK create \
    --driver=bridge \
    --subnet=192.168.41.0/24 \
    --gateway=192.168.41.1 \
    nos3_core
echo ""

echo "Launch GSW..."
$BASE_DIR/cfg/build/gsw_launch.sh
echo ""

echo "Create NOS interfaces..."
export GND_CFG_FILE="-f nos3-simulator.xml"
gnome-terminal --tab --title="NOS Terminal"      -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name "nos_terminal"        --network=nos3_core -w $SIM_BIN $DBOX ./nos3-single-simulator $GND_CFG_FILE stdio-terminal
gnome-terminal --tab --title="NOS UDP Terminal"  -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name "nos_udp_terminal"    --network=nos3_core -w $SIM_BIN $DBOX ./nos3-single-simulator $GND_CFG_FILE udp-terminal
echo ""

# Note only currently working with a single spacecraft
export SATNUM=1

#
# Spacecraft Loop
#
[... 70 lines cut ...]
    gnome-terminal --tab --title=$SC_NUM" - RW 2 Sim"     -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name $SC_NUM"_rw_sim2"      --network=$SC_NETNAME -w $SIM_BIN $DBOX ./nos3-single-simulator $SC_CFG_FILE generic-reactionwheel-sim2
    gnome-terminal --tab --title=$SC_NUM" - SAT_CAM Sim"  -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name $SC_NUM"_sat_cam_sim"  --network=$SC_NETNAME -w $SIM_BIN $DBOX ./nos3-single-simulator $SC_CFG_FILE sat_cam_sim
    
    gnome-terminal --tab --title=$SC_NUM" - Radio Sim"    -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name $SC_NUM"_radio_sim"    -h radio_sim --network=$SC_NETNAME --network-alias=radio_sim -w $SIM_BIN $DBOX ./nos3-single-simulator $SC_CFG_FILE generic_radio_sim
    
    #gnome-terminal --tab --title=$SC_NUM" - Sample Sim"   -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name $SC_NUM"_sample_sim"   --network=$SC_NETNAME -w $SIM_BIN $DBOX ./nos3-single-simulator $SC_CFG_FILE sample_sim
    gnome-terminal --tab --title=$SC_NUM" - StarTrk Sim"  -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name $SC_NUM"_startrk_sim"  --network=$SC_NETNAME -w $SIM_BIN $DBOX ./nos3-single-simulator $SC_CFG_FILE generic_star_tracker_sim
    gnome-terminal --tab --title=$SC_NUM" - Torquer Sim"  -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name $SC_NUM"_torquer_sim"  --network=$SC_NETNAME -w $SIM_BIN $DBOX ./nos3-single-simulator $SC_CFG_FILE generic_torquer_sim
    echo ""
done

echo "NOS Time Driver..."
sleep 8
gnome-terminal --tab --title="NOS Time Driver"   -- $DFLAGS -v $SIM_DIR:$SIM_DIR --name nos_time_driver --network=nos3_core -w $SIM_BIN $DBOX ./nos3-single-simulator $GND_CFG_FILE time
sleep 1
for (( i=1; i<=$SATNUM; i++ ))
do
    export SC_NUM="sc_"$i
    export SC_NETNAME="nos3_"$SC_NUM
    export TIMENAME=$SC_NUM"_nos_time_driver"
    while ! $DNETWORK connect --alias nos_time_driver $SC_NETNAME nos_time_driver; do echo "retry..." && sleep 3; done; echo "success..."
done
echo ""

echo "Docker launch script completed!"