nasa/nos3 #339: Reintroduce AIT

cjjacks, 2024-07-19:
Reintroduce AIT into the nos3 ecosystem as an available gsw.  Potentially include something other than AIT-GUI as the interface like openMCT or AstroUX.

zlynch2, 2024-07-25:
Hello,

to keep you updated we are looking into this issue. We will need to upgrade our AIT and will most likely add this option along side our fprime feature since fprime has its own  basic gds. I will keep you updated as we continue our work with NOS3 and adding different GSW to select from when using nos3.

cjjacks, 2024-07-25:
Zachary,

I've actually forked nos3 and started work on adding in AIT.  I am fairly
familiar with AIT as I have made some contributions there as well.  Is it
okay if I continue working on this with the aim to merge it back into nos3
in the future?  Or will this conflict with the work you are doing for
fprime?

Cameron

jlucas9, 2024-07-26:
@cjjacks that would be great! If you wanted to submit a pull request once you're ready and send in the contributor agreement (https://github.com/nasa/nos3?tab=readme-ov-file#contributions) we can pull it right in!

zlynch2, 2024-09-12:
Going to close this issue as we merged ait into its own branch nos3#339-AIT (this was the fork mentioned above, PR was approved)

jlucas9, 2024-09-13:
Still working this in the new branch related to this issue

jlucas9, 2024-10-16:
Merged into `dev`!

closed 2024-10-16 by jlucas9