nasa/nos3 #340: Unrealistic Data

gracekinney, 2024-07-22:
Hi everyone,

I am working on a project using NOS3 as the main focus for generating different satellite scenarios, where we will use this data for testing. My team and I have been able to get NOS3 running from the VM to an API, and we have successfully set-up the simulation so that the terminal commands make prep, and commands from Igniter: make clean, make, and make launch all work, and we are able to run an instance of the simulation.

However, we are running into an issue. We are not seeing realistic telemetry data from the simulator when we run different scenario commands. For example, running from the Command Sender: target GENERIC_EPS, command GENERIC_EPS_SWITCH_CC. We have found that when we use Command Sender to turn all switches ON, the BATT_VOLTAGE decreases, and when we turn all switches OFF the BATT_VOLTAGE decreases. We expect to see the opposite: switches ON BATT_VOLTAGE increases, switches OFF BATT_VOLTAGE decreases. We have theorized that the switches may be backwards?

Please refer to the pdf report I have attached, which highlights unrealistic data and includes screenshots from Telemetry Grapher and Packet Viewer.

The bottom line is that we are experiencing issues with the telemetry data not being realistic, and we believe NOS3 is much more capable than what we are experiencing.

I would like to ask the developers to help resolve the issue of unrealistic telemetry data, and potentially clarify how and why we are experiencing this.

Thank you!

[SatSim_EPS_demo_report_GK.pdf](https://github.com/user-attachments/files/16334015/SatSim_EPS_demo_report_GK.pdf)

Isgaroth-the-Green, 2024-07-23:
Good morning.

Just to clarify, are you saying that you currently find the voltage decreases both when the switches are turned off, and when they are turned on? Or that the voltage increases when the switches are off and decreases when they are on? The attached document seems to imply (unless I am misunderstanding it) that the voltage generally increases when switches are off and decreases when they are on, which seems correct behavior. The switches are to turn on or off additional power draws, so turning them all on - and thereby drawing more current - should be correlated with decreasing charge (and thus battery voltage). 

Regarding the other changes to the battery voltage, those are probably created by the direction of the simulated solar panels. We do not assume solar panels on every face of the satellite, so as it moves along its orbit it would be expected to go through periods of better and worse solar exposure. That may not be the only thing contributing to the unforced voltage changes, but it is at least a contributing factor.

gracekinney, 2024-07-24:
Hello,

My team and I re-ran the scenario of turning all switches off in GENERIC_EPS via Command Sender. We plotted BATT_VOLTAGE and SA_VOLTAGE and viewed the data via Telemetry Grapher and Packet Viewer. We were looking to see if anything outside of switch mnemonics had a response to our changes, and thought voltage would be our best bet to view a response. Our issue is that we do not see any other subsystem respond to turning all the switches off/ we do not know where to look to see if there is a response. 

Do you know if there is any documentation for how the subsystems interact/ respond when changes are made to certain subsystems? This information would be very helpful with determining what we need to change for creating scenarios!

Thank you!

closed 2024-10-24 by Isgaroth-the-Green