nasa/nos3 #344: Coloring in a custom .obj model

Matcha-jpg, 2024-07-29:
I was wondering how to color my custom spacecraft in the 42 cam simulation. I see that most of the Model .obj files are created with Wings 3D, but it doesn't seem to matter if the .obj files were created with Wings or not. I have imported a file from blender to utilize, along with its necessary .mtl file, but my spacecraft still doesn't show color. I have tried exporting one of the base models that does show off color (JustACube.obj) and changing its .mtl properties in Blender and imported the resulting .obj and .mtl file, but it still shows the same color from before the edits were made.

Matcha-jpg, 2024-07-29:
Never mind, further inspection of certain .mtl files revealed the issue.

closed 2024-07-29 by Matcha-jpg

shivanandanansac-creator, 2025-12-15:
How do I add my model to NOS3 ?