nasa/nos3 #345: Multiple Spacecraft Using Cosmos5

matt11matthew, 2024-07-30:
Are there any general best practices for having multiple space craft (with option to send commands/request telemetry) from individual craft?

Ravigd882353, 2024-08-26:
any update on this one ?

Isgaroth-the-Green, 2024-08-26:
About what, precisely, are you asking for best practices? For having multiple spacecraft that can communicate with each other? Or the differences between a single and multiple spacecraft? Or something else entirely? 

We are currently working on getting multiple spacecraft to be able to launch simultaneously, but we have not yet progressed to the point of having them communicate with one another (and I do not believe the multiple spacecraft functionality is yet in the main branch, either).

Isgaroth-the-Green, 2024-08-26:
As far as I know, the answers to your questions would be the following:

1. Switching from COSMOS4.5 to COSMOS5 (which, I think, is also called OpenC3) is as simple as changing "COSMOS" to "OpenC3" in the relevant file ([nos3-mission.xml](https://github.com/nasa/nos3/blob/main/cfg/nos3-mission.xml), in this case).
2. If you create a hardware model for another component, you can run it with as many satellites as NOS3 will support (still only 1, at the moment). There is not anything that would prevent a hardware model from being used for multiple satellites simultaneously; am I understanding your question correctly?
3. We are presently working on getting all the spacecraft to connect with their own instance of COSMOS ([nos3 issue 287](https://github.com/nasa/nos3/issues/287)). We do not yet have support for multiple spacecraft connected to the same single GSW. 
4. That is a very broad question; dependencies for NOS3 generally? Since everything is running in containers, the dependencies should come automatically. And what exactly do you mean by traceability? Our documentation, which will contain some of what you seem to be asking about, can be found at the wiki [here](https://github.com/nasa/nos3/wiki).

Ravigd882353, 2024-09-05:
Thank you for your insights & helps. it's really helpful...OpenC3 is working for me.  but i think, i am not more clear or other questions...let me put it this way...

2) 
As you can see in the screenshot that i attached...i am able to generate 3 satellites with it's orbit. but i don't understand that how i can connect hardware model with each of 3 satellites ?  and if it's connected, then how i can extract the data from it ? 
[image]

4) 
By Dependencies and traceability i mean,(in simpler term) that how i can find which file or variable has been used where  ? 
like as example...Inp_graphics.txt from cfg/inout...this file has been used nowhere but if i am changing the parameters of MAP and CAM,i am getting required result. 
so the question is how and where this file has been used ? and how i can get proper traceability of each file ? do you have any structure on that ? 

1) 
I want to calculate Az/El for simulated satellite. some of python files has been created  in nos3/gsw/OrbitInviewPowerPrediction/scripts/  for that. 
from my understanding it has been generated for TLE testing. but how can i get TLE  or Az/El  for satellite that i am simulating (not pre-defined satellites) ?

Isgaroth-the-Green, 2024-10-09:
Sorry about the delay in responding. At present, we do not have things set up to be able to easily run multiple satellites at the same time; that is an in-progress effort. In your case I would recommend creating three copies of COSMOS (one per satellite) to get the data; that is probably the only way to get the information you want (one hardware model and one copy of COSMOS per satellite). 

Regarding your dependencies and traceability question, I believe most IDEs have the ability to trace files, and (for my part) I will typically use grep in Linux to figure out where the parameters I am looking for might be. We will be working on more formal requirements and traceability in the next phase of the project, though. 

As regards calculating Az/El or TLE, I am not entirely sure how best to approach that. @marksuder might have some input, though.

msuder, 2024-10-09:
A TLE file input parameter can be set in the orbit file (Orb_LEO.txt or similar) on lines 13, 23, 24.  TLEs for known objects can be retrieved from https://www.space-track.org/; you need to create a free account to use that site.

jlucas9, 2025-03-26:
Closing due to age