nasa/nos3 #346: JSON API Cosmos 4.5

matt11matthew, 2024-07-31:
How would you enable the JSON API on Cosmos 4.5 since it tends to be more friendly to multiple spacecraft, whenever I run Cosmos 4.5 the API ports have no data.

jlucas9, 2024-08-01:
I haven't actually used the JSON API for COSMOS before unfortunately. @Isgaroth-the-Green or @kevincbruce any ideas?

matt11matthew, 2024-08-02:
I think if it was possible to create a more detailed guideline on implementing external applications using TCP, sending packets, and commands would be a major help to many people who don't have a large amount of experience on the software side of space. 

Since I know the project is moving away from Cosmos 5 that creates a major limitation for people who want to make there own frontend.

jlucas9, 2025-03-26:
Closing due to age

closed 2025-03-26 by jlucas9