nasa/nos3 #350: memory leak ?

gardinwin, 2024-08-09:
Hi.

After starting the simulator, the memory usage decreases over time. 
Should the messages or resources received by the user be released? 
Or does the automatic garbage collector periodically release the resources?

Thanks.

jlucas9, 2024-08-30:
Hi @gardinwin, 

Thanks for bringing this to our attention. After launching in my VM and getting to a steady sun pointed state (1026):

[image]

I just let it run for awhile (1105) and memory use from 42 appears to continue to increase:

[image]

I'll keep this issue open and continue to investigate to get this resolved.

gardinwin, 2024-08-31:
image.png (view on web)
image.png (view on web)

msuder, 2024-10-18:
I ran 42 under valgrind (leakcheck).  None of the findings turned out to be a memory leak.

msuder, 2024-12-18:
To run 42 under valgrind, you have to modify the docker container that runs 42.  This requires starting the container from an image, executing the commands in the container to install valgrind, and then saving the container as a new image.  This new image then needs used to run 42 under valgrind.

msuder, 2024-12-18:
I redouble checked today with `valgrind --leak-check=full` and did not see any obvious memory leaks.

msuder, 2024-12-18:
I ran the EPS nos3-single-simulator under valgrind (leakcheck) and it  showed no leaks.  I also watched htop and the simulators and  they were holding a constant amount of memory.  So the simulators are not leaking memory.

msuder, 2024-12-18:
This issue is recommended for closure.

closed 2024-12-19 by jlucas9