nasa/nos3 #351: Star Tracker's Quaternion seems to be off

lucapizzuto99, 2024-08-09:
Hi everyone,

I am developing an attitude mode to align a vector in ECI (N) frame with the b2 axis, similar to what the Sunsafe mode does.
To move from the inertial frame to the body frame I need to retrieve the inertial quaternion of the body, which should be provided by the Star Tracker in the `generic_adcs/fsw/src/generic_adcs_ingest.c` script as:

```
void Generic_ADCS_ingest_generic_st(CFE_MSG_Message_t * Msg, Generic_ADCS_DI_St_Tlm_Payload_t *St)
{
    GENERIC_STAR_TRACKER_Device_tlm_t *st = (GENERIC_STAR_TRACKER_Device_tlm_t *)Msg;

    St->valid = st->Generic_star_tracker.IsValid;
    double q[4] = {st->Generic_star_tracker.Q0, st->Generic_star_tracker.Q1, st->Generic_star_tracker.Q2, st->Generic_star_tracker.Q3};
    QxQ(q, St->qbs, St->q);
}
```
In particular, the QxQ method should move from the Sensor frame to the Body frame, by using the quaternion specified in  `generic_adcs/fsw/cfg/Inp_DI.txt` as :

```
!!!!!!!!!!!!!!! Star Tracker !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
-0.5510871 -0.4059919 -0.5108982 0.5200544 ! Quaternion from sensor to body frame
```

As far as I understand, then the resulting quaternion stored in the DI_St_Tlm_Payload data structure should be the body quaternion that I am looking for. However, when I confront this quaternion with the one directly retrieved from 42 in the 42Truth Sim, they are totally different. 
This leads to a wrong conversion between my ECI and body frame, while the conversion becomes correct when, instead of retrieving the ST quaternion, I retrieve directly the body quaternion from 42. 

Is there something that I am missing, or is perhaps the quaternion specified in Inp_DI the wrong one? 

Also, to get the body quaternion from 42, I had to modify the 42 code itself to return the ST quaternion as S->B[O].qn instead of the one computed by the method in the  [(42sensors.c script](https://github.com/ericstoneking/42/blob/master/Source/42sensors.c).
For some reason, modifying the data_provider.cpp code of the ST in this way:
```
append("SC[").append(std::to_string(_sc)).append("].AC.ST[").append(std::to_string(_st)).append("]"); // SC[N].AC.ST[M].qn / Valid
qn_key.append("SC[").append(std::to_string(_sc)).append("].B[0].qn"); 
std::string qn_values = _dp.get_value_for_key(qn_key);

_generic_star_tracker_data_is_valid = (valid_value == "1");
std::vector<double> data;
parse_double_vector(qn_values, data);
```
leads to the collapse of the ST sim.

Thanks in advance for the support, I really appreciate it!

jlucas9, 2024-10-18:
@marksuder can you check this one out?

msuder, 2024-12-17:
Change Inp_DI.txt and SC_NOS3.txt to both use the identity (no) rotation.  This causes the DI quaternion and the truth quaternion to track each other (when the star tracker is valid).

This issue is being closed and a new issue is being opened to get Inp_DI.txt and SC_NOS3.txt to have inverse, non-identity rotations for the star tracker.

msuder, 2024-12-17:
The  new issue to track fixing the inverse rotations is:  https://github.com/nasa/nos3/issues/431.

closed 2025-01-15 by kevincbruce