nasa/nos3 #359: CFDP SendFile with NOS3 latest release

lucapizzuto99, 2024-09-02:
Hi everyone,

I would like to use the CFDP protocol to exchange files within NOS3, between two different s/c. For some reason it seems like the Cosmos command `CFDP_DEBUG/SEND FILE` is not working. 

```
EVS Port1 42/1/CFE_SB 21: Send Err:Invalid MsgId(0x4346)in msg,App CI_LAB_APP
EVS Port1 42/1/CI_LAB_APP 10: CI: L367, CFE_SB_TransmitBuffer() failed, status=-905969661
```

I made sure the CMake files include CFDP and I tried to follow back the old commits from previous issues(ex. 95), but I still haven't figured it out. Also, the CMD file for CFDP looks a bit odd:
```
# Do not attempt to modify this file! May break the engine!

COMMAND CFDP SEND_FILE BIG_ENDIAN "Starts sending file with CFDP"
  APPEND_ID_PARAMETER FUNC_ID 72 UINT 0x434644505f53454e44 0x434644505f53454e44 0x434644505f53454e44 "CFDP SEND_FILE CODE"
  APPEND_PARAMETER CLASS 8 UINT 1 2 2 "CFDP Class 1 or 2"
  APPEND_PARAMETER DEST_ID 128 STRING "24" "Destination ID - 2 bytes"
  APPEND_PARAMETER SRCFILENAME 512 STRING "/media/sf_gtosat/gsw/scripts/launch.sh" "Path/Filename to send"
  APPEND_PARAMETER DSTFILENAME 512 STRING "/cf/launch.sh" "Path/Filename at destination"
```

If anyone is willing to help me to understand how to transfer files between s/c using NOS3's implementation of CFDP I would highly appreciate it!

Cheers,
Luca

jlucas9, 2024-10-17:
Hi @lucapizzuto99,
Apologies for the delay getting back to you. In the latest `dev` branch of NOS3 (and also likely in the latest 1.7.1 release) CFDP with COSMOS should be functional. Side note that I have reports that it is possible to crash out FSW when using CFDP, but haven't tracked it down yet to verify.
If you're actively running, you can open the "Test Runner" from the NOS3 Launcher of COSMOS. If you then navigate to Mission Test > LPT > test_cfdp_small_c1 and click start you will see that the RADIO interface of the Command and Telemetry server gets activated. This test then uploads a small test file to the spacecraft, then proceeds to download the same file from the spacecraft as well. There are other tests for Class 2 (think of this like TCP where retransmits of missed data can occur).

https://github.com/nasa-itc/gsw-cosmos/blob/a8d96bc426554cd09ee332e78ea4d5e3982f4623/config/targets/MISSION/procedures/mission_test.rb#L32

Please let the team know if you have any questions or other follow ups! Have a great day!

[image]

jlucas9, 2024-10-31:
Closing due to age - please reopen if still an issue!

closed 2024-10-31 by jlucas9