nasa/nos3 #361: RTS command ?

gardinwin, 2024-09-04:
Hi
I have a questions about new component.
Should I add a command in RTS related files ?
What modifications are necessary to perform RTS?
I would like to know the purpose of rts001.c and rts005.c.

jlucas9, 2024-10-18:
Hi @gardinwin,
You can certainly add additional or modify commands in the RTS files or even create new RTSs as you want. Note that RTS 1 is run during a power on reset from cFS while RTS 2 is the processor reset version. Other RTSs are not hard coded, but typically referenced from the limit checker (LC) application as a response to a watch point / action point being triggered.

As a quick example - if spacecraft telemetry reports error counters in the sample instrument above a set threshold, we could run and RTS to safely power cycle the instrument as an attempted correction.

Let us know if you have any other questions and have a great day!

jlucas9, 2024-10-31:
Closing due to age - please reopen if still an issue!

closed 2024-10-31 by jlucas9