nasa/nos3 #362: TM Transfer Frames, Issue with idle packet generation

Simon9050, 2024-09-04:
Hi, 

I am trying to use TC / TM Transfer Frames with NOS3.  The TC TF part works fine, but I have seen that there is currently an issue with the TM TF part due to Idle Packet generation ... 

[[nasa/nos3#202] Comment out idle packet generation due to it causing … · nasa-itc/CFS_TO@a3b534d (github.com)](https://github.com/nasa-itc/CFS_TO/commit/a3b534dedd2e212d0eb2a500b2eac67cd30c497a)

In particular, if I try to use TM TF in the simulator, after the TC for TO_ENABLE_OUTPUT, I receive an error on the FSW tab : "The Idle Packet has MsgId other than 0x3ff" ... 

Do you have the same issue on your side ? Are you working on some fix ? 

For info (hope this helps), I think that the origin of the issue is the function TM_SDLP_AddData (called by the function TM_SDLP_SetOidFrame, line 600) :

[CFS_IO_LIB/fsw/src/services/tm_sdlp.c at 523edf70f8a620f4a7988eef3beb23713289d1a4 · nasa-itc/CFS_IO_LIB (github.com)](https://github.com/nasa-itc/CFS_IO_LIB/blob/523edf70f8a620f4a7988eef3beb23713289d1a4/fsw/src/services/tm_sdlp.c)

Indeed, while running the simulator, I can see that the MsgId of the pIdlePacket is 0x3ff before  iStatus = TM_SDLP_AddData line 600, but it is modified right after (leading to the error message line 423).

I think I will keep working on this, but If you have some suggestions or a simple fix, I am very interested.  

Thanks and regards,

Simon

Simon9050, 2024-09-06:
Update :  Please note that the pointer arithmetic of line 760 in the function TM_SDLP_AddData is not correct. 

    CFE_PSP_MemCpy((void *) (pFrameInfo->frame + pFrameInfo->currentDataOffset),
                   pData, lengthToCopy);

A simple fix could be : 

    CFE_PSP_MemCpy((void *) ((char *) (pFrameInfo->frame) + pFrameInfo->currentDataOffset),
                   pData, lengthToCopy);

Please let me know if this works for you, 

I still have some issues on the TM TF part... so if you have some suggestions or fix to be released please keep me posted.

Thanks and regards, 

Simon

jlucas9, 2024-10-18:
Hi @Simon9050,
I can confirm I've seen these issues before. @rjbrown2 / @dccutrig could you two take a look and confirm Simon's issue and possible fix?

Thanks for reporting this!

dccutrig, 2025-04-03:
Internal note to self to compare io_lib from another project to diff for changes.