nasa/nos3 #364: Control Satellite Attitude ?

gardinwin, 2024-09-11:
Hi
I'd like to control attitude.
To change the satellite's attitude to Right Looking or Left Looking for mission operations, what command can be used to control it?

Thanks.

gardinwin, 2024-09-11:
SC_Simle.txt contains : 
*************************** Initial Attitude ***************************
 13 NAN                         ! Ang Vel wrt [NL], Att [QA] wrt [NLEF]
 14 0.3    0.4    0.5           ! Ang Vel (deg/sec)
 15 0.0    0.0    0.0    1.0    ! Quaternion
 16 20.0    30.0    40.0    213    ! Angles (deg) & Euler Sequence

Is it necessary to adjust the satellite to point directly downward toward the Earth?

jlucas9, 2024-10-18:
Hi @gardinwin,
Presently the generic_adcs utilized in NOS3 only has a sun pointing mode. Issue #181 has the goal to extend this to allow for 3 axis stabilization / pointing. This is in the near-term roadmap to knock out.
Please let us know if you have other questions, issues, or requests we can add to our roadmap!

jlucas9, 2024-10-31:
Closing due to age - please reopen if still an issue!

closed 2024-10-31 by jlucas9