nasa/nos3 #367: YAMCS Studio Integration

jlucas9, 2024-09-24:
Integrate YAMCS studio into NOS3 (if not already a part of initial YAMCS)

rjbrown2, 2024-11-21:
YAMCS Studio Integration Instructions:
Obtain the tarball appropriate to your system.  Note that YAMCS Studio requires JRE for Java 11 or higher.
https://github.com/yamcs/yamcs-studio/releases

Extract the tarball to your system, and you can run then ./'Yamcs Studio'
The YAMCS studio User's Guide is a great resource for setting up these interfaces, and this is how the JSTAR team accomplished this.  
We've also created a custom OPI file for the SAMPLE application that will allow users to view how scripting works to connect buttons to commands and telemetry.

Users can create a new project within the Display Builder within YAMCS Studio, and then import JSTAR's NOS_SAMPLE_APP.opi
Select the OPI file, and click the green play button to advance to the Display Runner (This will bring up the NOS Sample App OPI window)
From the Options ribbon at the top, select YAMCS, and then Connect.  (This assumes you already have NOS3 and the YAMCS Web gui running, functional, and receiving telemetry).
[image]

Following successful connection, users should be able to run the Enable_CC command from the sample app OPI, and begin seeing telemetry:
[image]

The OPI File will be uploaded as part of the #80 upcoming merge into dev.

rjbrown2, 2024-11-21:
OPI File lives within:
gsw/yamcs-studio/  Under the NOS3 root directory.

closed 2024-11-21 by rjbrown2