nasa/nos3 #376: fprime merge cleanup from nos3#84

zlynch2, 2024-10-10:
- [ ] Review Star tracker config (uart number) in nos3-simulator.xml.
- [ ] Test 42 provider with star tracker, in nos3-simulator.xml
- [ ] Should we have a single launch script and then calls the various pieces based on configuration instead of two separate ones to manage? (cfs_cosmos.sh)
- [ ] docker_debu.sh, need to review and make docker debug. ports needed for fprime?
- [ ] Configure.py, Should we overwrite the ./scripts directory or move this into the hidden ~/.nos3 or a build directory and run it out of there like we do for GSW?
- [ ] Docker_checkout.sh, need multiple docker checkouts depending on FSW running fprime, cfs, etc.
- [ ] F' gds running direct in terminal changed to its own, and fprime fsw launching in its separate terminal

zlynch2, 2024-10-15:
merge from @jlucas9 feature branch fixed f' gds and f' fsw launching in own windows.

jsilveira1409, 2024-11-01:
hi @zlynch2 ! i would love to contribute to this integration of fprime with nos3, let me know how I can help

zlynch2, 2024-11-04:
hi @jsilveira1409 that would be awesome, first we would need you to return some paper work in order for you to get started. Ill have @jlucas9 get that to you.

zlynch2, 2024-11-04:
Acutally @jsilveira1409 looks like you can find what you need here: 
https://github.com/nasa/nos3?tab=readme-ov-file#contributions

Follow those steps, and you can start contributing to fprime work

zlynch2, 2024-11-14:
Star tracker is being fixed with correct uart in #351

zlynch2, 2024-11-14:
Docker checkout will be added to development roadmap

zlynch2, 2024-11-20:
Closing issue, above issues tracked and committed in other issues. Uart 10 fixes star tracker, and 42 provider

closed 2024-11-20 by zlynch2