nasa/nos3 #387: Processor emulation options

jlucas9, 2024-10-16:
Looking to assess all the options to migrate from cross compiled to processor emulation:
* QEMU
* Renode.io
* ?

dgeletko, 2024-10-29:
I looked into unicorn a while ago and it seems like a great project, but keep in mind they basically stripped everything out of qemu other than the CPU models, so there are no devices. It seems to be meant as a core framework to build other tooling, plus it has numerous language bindings. It would be great if we wanted to build our own product _almost_ from scratch (which I love the idea of), but for what we do it seems like it would be way too much effort given our team size and priorities. We might be able to use it as a starting point and pull qemu devices back in (PCI, timers, etc.) which could essentially give us a qemu 'lite', but not sure that would buy much other than removing all of the non-modeling code in qemu. Also keep in mind it is also a GPL2 license.

dgeletko, 2024-10-29:
I haven't heard of Qiling before but it looks interesting, however, based on the docs it is more comparable to qemu user mode emulation, where you can execute a binary compiled for a different architecture, which is different than the system emulation we do where we execute the binary on an emulated system of the target architecture.

jlucas9, 2024-11-14:
QEMU

jlucas9, 2025-01-31:
https://github.com/rust-vmm/vhost-device

closed 2025-02-12 by jlucas9