nasa/nos3 #412: Spacecraft telemetry not behaving as expected

BaptisteRC, 2024-11-19:
### Discussed in https://github.com/nasa/nos3/discussions/411

<div type='discussions-op-text'>

<sup>Originally posted by **BaptisteRC** November 15, 2024</sup>
Simulating a 500km SSO satellite, I would expect the various subsystem telemetry to vary with the solar illumination/eclipse.
Instead, I have noticed that:

- All 3 temperature endpoints have constant values (RAW_BATTERY_TEMPERATURE, RAW_EPS_TEMPERATURE, RAW_SA_TEMPERATURE)
- SA voltage is constant across multiple orbits (RAW_SA_VOLTAGE)
- BATV (RAW_BATTERY_VOLTAGE) is constantly declining, until a step change to recover then it declines again at same rate… (unrelated to the eclipse time or satellite attitude)
- ADCS telemetry data seem to vary ok at first glance (CSS_SVB vary with attitude & eclipse, Angular_Acceleration...)
I should also mention that the CAM42 and other specific ADCS parameters confirms I am in eclipse.

Am I misunderstanding the use case of NOS3 as real time simulator?
Is there an additional configuration I have missed or do I need to develop my own software to obtain a more realistic subsystem telemetry simulation?

Thanks much!

BAT & SA VOLT chart:
[image]
TEMP chart:
[image]
</div>

Isgaroth-the-Green, 2024-11-21:
Good morning.

As of right now, we do not have any variation in the temperatures programmed in; that is (as you point out) not accurate to reality, but it is a component of the simulation we have not yet gotten to. 

The battery voltage decreases with decreasing battery charge; right now, I do not believe there is any checking of whether the satellite is in eclipse or not, just checking the angle of incidence between the sun vector and each individual solar panel.  The reason for the general decrease of the voltage is that the system as we have it set up runs a slight power deficit; there should be limits added (to keep from overflowing in the manner you saw with the jump discontinuity) but that is, as with the temperature, something we have not yet gotten modeled.  The SA_VOLTAGE is in a similar state; its value varies within the simulator components, but never finds its way into the device and app files which are accessible via telemetry. 

You are not misunderstanding the use case of NOS3; it is intended to operate as a real-time simulator but at present has much greater fidelity in terms of software and communications across the various buses than in its relation to the space environment itself. 

Incidentally, you may have already seen this, but we have some high-level documentation in the [wiki](https://github.com/nasa/nos3/wiki).  We are planning to further improve the documentation.

BaptisteRC, 2024-11-21:
Thanks for the reply!
We'll wait a bit longer then.

closed 2024-11-21 by BaptisteRC