nasa/nos3 #413: Errors during ARDUCAM picture aquisition process and no change in TLM

BaptisteRC, 2024-11-19:
I try to use the ARDUCAM subsystem to simulate taking a picture.
In the absence of user manual, I only guess the proper process. I have noted a few issues detailed below.

1. Picture errors on first CMD
When triggering the command CAM_EXP3 for the first time, I get the below error messages in the NOS3 debug terminal:
[image]
I understand those are potentially related to the bus and the FSS / MAG subsystems, however it might have an impact on my CAM telemetry..
Q: is that expected / of nature of cause an issue later on e.g. receiving CAM telemetry?

3. Picture telemetry:
I dont see the command counter increases (/ARDUCAM/ARDUCAM_HK_TLM_T/COMMANDCOUNT), and the other telemetry points (like CAM_DATA) are not refreshed either.
Q: is that expected? (do we need to code that part ourselves?)
Q: If not expected, how do we downlink the image data/file, if it exists?
Ultimately I'd be fine with just a dummy file on the file system I can then download...
[image]

3. Picture acquisition time & size
It takes about 6 minutes until the CAM EXP3 is "complete" as per NOS3 debug window. 
Q: Is that timeline expected?
From the code, it seems there are 3 commands for 3 size of picture CAM_EXP1, CAM_EXP2, CAM_EXP3.
However in COSMOS gsw only CAM_EXP3 is available (which is the "large" picture command).
Q: is that expected? is there a way to take smaller pictures?

Thanks much for your help.

BaptisteRC, 2024-11-21:
Thank you again!
RE Q2, the commandcounter issue might be specific to my setup.

In your setup, do you receive multiple CAM_DATA packets and do you use those to extract a picture? or it is random dummy data?

Is my understanding correct that the CAM_EXP3 command doesnt trigger picture file creation, only cam_data telemetry packets to be broadcasted immediately?

Thanks much!

Isgaroth-the-Green, 2024-12-05:
The data are not random dummy data; they are supposed to create a picture.  CAM_EXP3 does not create a picture file, but sending that command immediately starts transmitting an existing picture via the multiple data packets (at least, that is my understanding of things).

jlucas9, 2025-04-10:
Closing due to age. Will be reconfirming that Arducam is functional in issue #578 and have system tests to reference!

closed 2025-04-10 by jlucas9