nasa/nos3 #418: Cant enable telemetry

KevinZheng2025, 2024-11-25:
I am trying to get GPS coordinates off of the simulations and cannot enable telemetry. I'm not sure if this is a bug or if I am not setting up something correctly.
[image]
[image]

This is what the guide is doing 
[image]

Isgaroth-the-Green, 2024-12-05:
Good afternoon.

Unfortunately, the guide is a little outdated; when running with COSMOS (as it appears you are) the way to get GPS coordinates should be to enable the NOVATEL GPS (which will be an option on the Command Sender dropdown menu).  There will be an "enable" command, but not an "enable_telemetry" command; enabling the device is assumed to be enabling telemetry, as it sends housekeeping (HK) data automatically.  

Two further questions: One, which version/branch are you running with?  Secondly, are you able to get any telemetry data from any of the components?

KevinZheng2025, 2024-12-09:
Thank you for your assistance. When I search up Novatel, I encounter two options for enabling the connection. However, when trying to establish a Telnet connection to the GPS port (which I believe is 4245), I am still experiencing issues with connecting to the server.

I am currently running on the main branch of NOS3 and I am also unable to get data from the other components.

[image]
[image]
[image]

Isgaroth-the-Green, 2024-12-10:
I see.  I would expect telnet not to work unless you have changed the networking setup; we are now using Docker and therefore can no longer connect to things just by using the loopback address.  What does your COSMOS "Command and Telemetry Server" look like?  Does it show all four interfaces connected, and if so, is data coming through on any or all of them?  
[image] is what it looks like (although there should be data coming through on the 'DEBUG' interface, and possibly on the 'RADIO' as well).

jlucas9, 2025-02-12:
Closing due to age. Please reopen if issue persists!

closed 2025-02-12 by jlucas9