nasa/nos3 #422: Sample App Coverage Test Improvement

kevincbruce, 2024-12-11:
Build of [nasa/nos3#407] and the information it'll provide to test utilizing that CodeCov data to improve and expand coverage tests to reach 100% coverage on an app, to serve as an example for future app coverage tests going forwards.

msuder, 2025-02-04:
.github/workflows/build.yml
make debug
make build-test
make test-fsw
mkdir -p doc/coverage
gcovr --xml-pretty -o doc/coverage/coverage_report.xml
gcovr --html --html-details -o doc/coverage/coverage_report.html
firefox doc/coverage/coverage_report.html

kevincbruce, 2025-03-25:
2 Follow on Issues:

Device Coverage for Sample

Coverage Reporting Script (Script comment above and all test commands for ease)

closed 2025-04-10 by jlucas9