nasa/nos3 #435: 42 support multiple satellites ?

gardinwin, 2024-12-19:
Hi
I want to change the NUM_SAT in docker_launch.sh to = n to run with multiple satellites. I am including 42 within the loop; can 42 be operated by launching only one on a separate computer?

Please let me know.

jlucas9, 2024-12-19:
Hi @gardinwin,
42 certainly supports multiple spacecraft. You would need to ensure that each spacecraft simulator launched looks at the correct sc[#] in 42 it is related to as well as commands. For simplicity in design and worried that multiple spacecraft in a single instance would cause further performance impacts we simply copied the environment from spacecraft to spacecraft wholesale.

Let us know if you have any other questions or something above doesn't make sense!

gardinwin, 2024-12-20:
Thank you for your response.
I would like to configure 42 as shown in the link below.
The current setup is running on Linux. I am looking for reference materials, configurations, or a shell script.

https://github.com/nasa/nos3/wiki/NOS3-Architecture : basic docker architecture

gardinwin, 2024-12-20:
In 42/42exec.c call InitInterProcessComm() listen port with Inp_IPC.txt
Just listen once.
Is there any other versions to support multiple client ?