nasa/nos3 #513: Find an Alternative to gnome-terminal that can work with more desktop environments

kevincbruce, 2025-02-20:
Find if we can launch terminals with a command other than gnome-terminal for compatibility with MATE, XTCE, KDE, Mac, WSL, etc.

kevincbruce, 2025-02-21:
Looks like `x-terminal-emulator`, if installed, provides a symlink to launch whatever the default terminal for your desktop environment is (so, gnome-terminal in base ubuntu, mate-terminal in mate, xterm in xtce, konsole in kde). However, it may be a debianism, so probably wouldn't work for RHEL/Fedora/Rocky, or for Mac.

kevincbruce, 2025-02-24:
Possible Mac Solutions - may need separate Linux vs Mac scripts if we want it to run native on Mac. https://superuser.com/questions/174576/opening-a-new-terminal-from-the-command-line-and-running-a-command-on-mac-os-x

kevincbruce, 2025-02-26:
May be relevant for MacOS: https://stackoverflow.com/questions/60412920/macos-script-to-open-terminal-with-multiple-tabs

haisamido, 2025-02-26:
It would be good to parameterize the invocation of the terminal so it can be overridden by the user.

kevincbruce, 2025-02-26:
Agreed. That is more or less what I am thinking. Have a parameter that can be auto-set using uname -s and command -v, or that can be manually set by user. Simply want to have a degree of automation if possible, for less technical users.

haisamido, 2025-08-01:
No need to do this if docker compose is used i.e. #772