nasa/nos3 #521: 💡 NOS3 / F' Time sync - does it control the schedule too?

ephraim71, 2025-02-27:
### Summary

@rjbrown2 / @zlynch2
Have a question about the NOS3/F' time sync, let me know if this is the right place to post. Also disclaimer that I'm new to both frameworks 😅

So if I understand correctly, in F', the time source will come from NOS3 : https://github.com/nasa-itc/fprime-nos3/blob/nos3-dev/Components/Nos3Time/Nos3Time.cpp

Now in F', lets say we have a few queued or active components which are driven by a Rate Group Driver : https://fprime.jpl.nasa.gov/latest/docs/user-manual/design-patterns/rate-group/

Right now, this Rate Group Driver is triggered every 1 second : 
https://github.com/nasa-itc/fprime-nos3/blob/676680f6a8218e04d704d08f4bf03f2f6498ab51/SampleSimDeployment/Main.cpp
this uses `SampleSimDeployment::startSimulatedCycle(1000);`
which calls 

```
void startSimulatedCycle(U32 milliseconds) {
    cycleLock.lock();
    bool cycling = cycleFlag;
    cycleLock.unLock();

    // Main loop
    while (cycling) {
        SampleSimDeployment::blockDrv.callIsr();
        Os::Task::delay(milliseconds);

        cycleLock.lock();
        cycling = cycleFlag;
        cycleLock.unLock();
    }
}

```
I assume Os::Task::delay(milliseconds) depends on the system time of where F' is running.   

Now imagine we are running a simulation at 2x speed - the time would update 2x faster, but wouldn't the schedule on F' still run at 1x speed? Shouldn't NOS3 also control how fast the rate group driver runs? 

### Use Case

F' also runs at same rate as simulation.

zlynch2, 2025-02-27:
Hi @ephraim71 ,

We override the time tick in F' by hooking into our NOS engine time tick. You will notice that if you speed up the NOS3 Simulation time, fprime will actually do the same. Nos engine is controlling the the tick to fprime so fprime's overall time is based on NOS3 Time. 

I believe you can adjust rate groups in fprime like you mentioned, however I think the rate groups in fprime will be synced to the time tick of NOS engine respectively as we increase or decrease simulation time. So basically if you speed up NOS3 Simulation time, fprime will also speed up. At least that was the thought when overriding the fprime default time for time synchronization with NOS3.  

Disclaimer, as of now we only have active components defined, so I would need too experiment with queued components or passive components in fprime to say definitively in our environment. if you see any inconsistencies feel free to let us know. If you have individual questions about the fprime rate groups, I think the fprime folks can answer some more in depth questions about fprime as well.

However I think the above describes our goal by driving fprime with our NOS time Tick from NOS engine

ephraim71, 2025-02-27:
@zlynch2 

`however I think the rate groups in fprime will be synced to the time tick of NOS engine respectively as we increase or decrease simulation time. `

I was going through how Os::Task::delay(milliseconds) worked (this is used to drive the rate groups). 
Here's the implementation : `fprime/Os/Posix/Task.cpp`         
```
Os::Task::Status PosixTask::_delay(Fw::TimeInterval interval) {
        Os::Task::Status task_status = Os::Task::OP_OK;
        timespec sleep_interval;
        sleep_interval.tv_sec = interval.getSeconds();
        sleep_interval.tv_nsec = interval.getUSeconds() * 1000;

        timespec remaining_interval;
        remaining_interval.tv_sec = 0;
        remaining_interval.tv_nsec = 0;

        while (true) {
            PlatformIntType status = nanosleep(&sleep_interval, &remaining_interval);
            // Success, return ok
            if (0 == status) {
                break;
            }
            // Interrupted, reset sleep and iterate
            else if (EINTR == errno) {
                sleep_interval = remaining_interval;
                continue;
            }
            // Anything else is an error
            else {
                task_status = Os::Task::Status::DELAY_ERROR;
                break;
            }
        }
        return task_status;
    }
```
Now, nanosleep() is a linux function, this depends on the *system* time and not *nos3* time, right? Doesn't this mean rategroups don't care about nos3 time to determine when to start a component according to schedule?

zlynch2, 2025-02-27:
Yes I think you are correct that the nanosleep() will use wall clock time. however in our system, nos engine is driving things forward each tick. So this means nos engine will always wait for fprime to do its thing before ticking forward. thus time sync between our Sim and FSW. 

Now, I personally would have to dive deeper into fprime rate groups to give you a more definitive answer. However the fprime folks have a discussion board for questions and I think it is pretty active. if you have specific questions about the fprime rate groups how they drive with a redefined time I would encourage you to look there (https://github.com/nasa/fprime/discussions).

I also know fprime I think uses its PosixTime component by default, which we replaced it with our nos3time component. I hope that gives some clarity on what we are doing with nos3 time with fprime and NOS3 as whole.

ephraim71, 2025-03-03:
@zlynch2 Thanks for the information. I'll try running an example deployment and get back on this.

zlynch2, 2025-03-05:
Closing for now, if issues arise later feel free to post here

closed 2025-03-05 by zlynch2

ephraim71, 2025-03-24:
@zlynch2 Got some time to re-run this.     
So here's what I did : 
In `nos3-simulator.xml` , I changed the sim-microseconds-per-tick so that the simulation runs 2x faster than realtime : 
```
<sim-microseconds-per-tick>20000</sim-microseconds-per-tick>
<real-microseconds-per-tick>10000</real-microseconds-per-tick>
```
To check how fast Fprime scheduler is running, I took a look at BufferManager, which is connected to rateGroup3 :      
`rateGroup3.RateGroupMemberOut[2] -> bufferManager.schedIn`             
Added a print statement inside `BufferManager's schedIn_handler` to check how fast it is running.          

```
auto now = std::chrono::system_clock::now();
auto duration = now.time_since_epoch();

         // Convert to seconds and nanoseconds
auto seconds = std::chrono::duration_cast<std::chrono::seconds>(duration);
auto nanoseconds = std::chrono::duration_cast<std::chrono::nanoseconds>(duration) -
                            std::chrono::duration_cast<std::chrono::nanoseconds>(seconds);

         // Convert to time_t for readable format
std::time_t current_time = seconds.count();
std::tm local_time = *std::localtime(&current_time);

         // Print time with nanoseconds
std::cout << std::put_time(&local_time, "%Y-%m-%d %H:%M:%S") << "."< std::setfill('0') << std::setw(9) << nanoseconds.count() << std::endl; 
```

Since rateGroup3 is 0.25Hz, this is expected to run every 4 seconds. 
However since we are doing a simulation 2x realtime, this is expected to run every 2 seconds.      
Now even when I'm running 2x realtime, this Buffer Manager is still running only every 4 seconds.          

```
2025-03-24 16:38:30.443320085
2025-03-24 16:38:34.443648800
2025-03-24 16:38:38.444110302
2025-03-24 16:38:42.444519186
2025-03-24 16:38:46.444966306
2025-03-24 16:38:50.445419709
```

Could you review the above steps?

zlynch2, 2025-03-25:
Like mentioned above, Nos engine is responsible for time sync. So difference here is we send a command from fprime, that goes to command our NOS Sims like sample, executes a command, and then goes back to fprime. If you notice your time in fprime gds youll notice it matches. Conceptually lets say this process and time sync is the wall clock time in fprime (since we are defining it that way with our posix time component). 

Now you are speaking about rate group time, time that is unique to fprime processing, and that is controlled by the Blockdrv (block driver) in fprime. This rate group or interval time could be set arbitrarily like you are saying to match how fast rate groups execute in fprime. 

But again thats more an Fprime question. If anything for NOS this could be something to watch for performance but we havent run into issues thus far. I hope this helps clarifying between the concepts of interval time and wall clock time with fprime and NOS3.

ephraim71, 2025-03-25:
Also if you have the time, could we hop on a quick Google meet to discuss this?

zlynch2, 2025-03-26:
With Simulation time We are simply increasing the rate at which we progress through the simulated mission as a whole as well as interacting with 42 and our simulations at that speed. 

After looking closer. yes, if you wanted to sync the rate at which we are driving fprime processing faster/slower/stopped, it looks like the block driver interacting with our rate groups may be where we need to set that. It looks like you are correct with respect to how fprime is processing. It looks like if we sync the block driver, processing may stop entirely as well not just the time the fprime deployment is utilizing for synchronization.  

You mentioned the timer tick and yes that is exactly what we are doing. With our posix time component we are essentially piggy backing off what the psp is doing. So in our fprime deployment which is running fprime, time is set to our nos3 posix time for our deployment.

So in Fprime, you'll see the time ticking and the channels updating at the same speed when you command. For example if you watch your channels in fprime an pause time with the nos time driver, youll see time in fprime stop. If may be a decimal off but this is simply firefox updating. 

Then, when you unpause time, fprime will continue ticking. The caveat here is that fprime is still its own process running even with the deployment paused. So you just wont get any new data from our nos sims until you unpause time. 

What you are speaking to with rate groups I beleive may effect the performance of the sim as well, but it may not effect the time at which commanding occurs since ultimately we define/set when we send commands and time stamp the data with our posix time driving our fprime deployment. I would need to experiment with the rate groups to see the performance between syncing however, currently we havent seen any large performance issues. This may be something to watch as we progress our integration. 

I would encourage you to talk to the Fprime FSW team if you have more questions about the rate groups in fprime. However I think I described how we are integrating with NOS. Like I said this may be something we watch as we continue integration, which would require modifying our blockDrv cycles with our nos engine tick.

ephraim71, 2025-03-26:
Thanks @zlynch2 

> After looking closer. yes, if you wanted to sync the rate at which we are driving fprime processing faster/slower/stopped, it looks like the block driver interacting with our rate groups may be where we need to set that. 

Could we reopen this issue so that we can use this to track the block driver sync?